Pulse-Eight/libcec #522: cec-tray 6.0.0 does not launch on Windows

nickkossolapov, 2020-07-10:
After installing libCEC 6.0.0, with default installation setup, the application doesn't launch. This is on a machine with Windows 10, after uninstalling libCEC 4.0.3.

The error `Object reference is not set to an instance of an object` comes up immediately when opening cec-tray.

[image]

Additionally, the tray icon is visible, but when trying to open the menu from the tray icon, another error occurs. 

[image]

```See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at LibCECTray.settings.CECSetting.Load()
   at LibCECTray.settings.CECSettings.get_OverridePhysicalAddress()
   at LibCECTray.ui.CECTray.InitialiseSettingsComponent(CECSettings settings)
   at LibCECTray.controller.CECController.InitialiseGui()
   at LibCECTray.controller.CECController..ctor(CECTray gui)
   at LibCECTray.ui.CECTray.get_Controller()
   at LibCECTray.ui.CECTray.TsAdvancedClick(Object sender, EventArgs e)
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
cec-tray
    Assembly Version: 5.0.0.0
    Win32 Version: 5.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Pulse-Eight/USB-CEC%20Adapter/x64/netfx/cec-tray.exe
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4001.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4150.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4180.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.3752.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
LibCecSharp
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Pulse-Eight/USB-CEC%20Adapter/x64/netfx/LibCecSharp.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.```

opdenkamp, 2020-07-10:
hmm odd, I've tried this on 2 different Windows 10 machines before I published it. I'll take a look it this, thanks

nickkossolapov, 2020-07-10:
I also tried it on a separate machine that I use for .NET Framework development, and I encountered the same issues, so I don't know if it's a dependency issue. The first machine is a fairly clean HTPC.

Maybe it's related to the exe that was published?

opdenkamp, 2020-07-10:
I've installed that exe on the development host and on another, and didn't see this issue.
Since you do .net development, could you try to pull the latest sources and build it on your host, to check whether that works: https://github.com/Pulse-Eight/libcec/blob/master/windows/create-installer.cmd

nickkossolapov, 2020-07-10:
When manual build the installed, that produced the same error on my computer. However, I think I found the error:

[image]

I've crated a PR which would fix this issue. With this fix, the application runs fine and detects my device correctly. There was a small issue that I noticed with the app, which I outlined in my PR. 

https://github.com/Pulse-Eight/cec-dotnet/pull/12

opdenkamp, 2020-07-10:
thanks, I've merged the fix and I'll pull it in here in the next build. I also noticed that there's still a hardcoded version number in the tray app when I looked at your debug dump that I'll fix :)

commit 7c535311 (2020-07-13): fixed: cec-tray didn't start in some cases. set the correct version

closes #522
closed 2020-07-13 by opdenkamp