Pulse-Eight/libcec #526: cec-tray crashes on start (Windows 10) after update to 4.0.7

marcelmah, 2020-07-30:
Hi,

Just updated from 4.0.5 to 4.0.7 but when I start cec-tray I immediatley get this error:
[image]

Version 4.0.5 started without a problem.

opdenkamp, 2020-07-31:
looks like it's using the old LibCecSharp.dll library, which is not compatible with the newer cec-tray
try installing libCEC 6.0.2 and, if you installed 4.0.7 because of Kodi, keep "Kodi add-on" unchecked in the installer (default)

opdenkamp, 2020-07-31:
if it still complains about the same error, then there's probably a rogue LibCecSharp.dll in your search path

marcelmah, 2020-07-31:
Hi Lars,

Updating to 6.0.2 fixed the issue, I can now start cec-tray.
I don't really need Kodi integration anyway, I need it to integrate in Windows.

Still looking for a way to have the windows sound volume up / down and mute send CEC commands by default.
I have a [RF remote](https://nl.aliexpress.com/item/33022881641.html?spm=a2g0s.9042311.0.0.418e4c4dEunzTG) to control Kodi, but the power and volume up / down and mute buttons are interpreted by Windows. So now I have to switch the remote to IR mode and then use the buttons aimed on my TV which will use CEC to control the volume on my Onkyo receiver.

opdenkamp, 2020-07-31:
cec-tray can't do this, but you could use EventGhost to set up custom commands: http://www.eventghost.net/
after installing eventghost, run libCEC's installer another time and select the EventGhost plugin.
not sure how to suppress volume control in Windows. perhaps change the default output device to another and let the onkyo handle volume

marcelmah, 2020-07-31:
Hmm to bad, because I can control the volume up / down and mute from the test tab of the cec-tray application.
Will look into EventGhost.

opdenkamp, 2020-07-31:
I may have hooked it up on the "foreground application" tab, not sure. I can't check it on this PC right now :)

marcelmah, 2020-07-31:
EventGhost loading of the plugin also fails:
Error loading plugin file: C:\ProgramData\EventGhost\plugins\PulseEight
Traceback (most recent call last) (0.5.0-rc6):
  File "C:\ProgramData\EventGhost\plugins\PulseEight\__init__.py", line 80, in <module>
    from cec_classes import UserControlCodes, CECAdapter, AdapterError # NOQA
  File "C:\ProgramData\EventGhost\plugins\PulseEight\cec_classes.py", line 53, in <module>
    from . import cec
ImportError: cannot import name cec

opdenkamp, 2020-07-31:
I'll have to look into this later, thanks

alturismo, 2021-02-12:
Hi, this seems to be still a open issue ?

have the same behaviour here, may possible as the cec file from the addon for eventghost is an "empty" dummy file ?

[image]

woobins, 2023-04-14:
> Hi, this seems to be still a open issue ?
> 
> have the same behaviour here, may possible as the cec file from the addon for eventghost is an "empty" dummy file ?
> 
> [image]

I'm running into the same issue.  Did this ever get resolved?

TDL-jp, 2023-06-18:
> > Hi, this seems to be still a open issue ?
> > have the same behaviour here, may possible as the cec file from the addon for eventghost is an "empty" dummy file ?
> > [image]
> 
> I'm running into the same issue. Did this ever get resolved?

Nope, as of 6-2023 no changes. The solution may be to install an eventghost plugin from eventghost's forums, but that site seems to be dead and the wayback machine is the only way to see it. Regardless, without the plugin from eventghost's creator, I don't see if there's any way to fix this.

closed 2025-02-27 by github-actions[bot]