Pulse-Eight/libcec #529: What is the difference between 6.0.+ and 4.0.+?

marshevms, 2020-08-25:
I am using manjaro with a raspberry pi 4. cec-client with libсeс 6.0. + can't find tv. The error that comes out:
`no serial port given. trying autodetect: FAILED`
But with 4.0.7 everything works fine. Because of this reason, apparently Kodi cannot be found by my TV either.

alecsandes, 2020-08-28:
Hello, I got the same problem ...
I have tried to remove the version 6 which I compiled from git. I have deleted the folders to which I downloaded the binaries.
I am learning still in this linux/raspbian enviorment  ... so i don't know what I did. 

Now every time I reboot Raspberry I start my tv :))
I then tried to apt-get remove the 4,0 version ... 

Maybe I will find a solution here to remove the 6th version and install cec-utils through apt-get as I don't want to erase my raspberry and start over...

Thanks!

marshevms, 2020-08-28:
From your answer, I assume you are using a debian based operating system. I talked to the manjaro linux maintainers, they said that the next version of kodi will use the libcec version that comes with kodi.

alecsandes, 2020-08-28:
> From your answer, I assume you are using a debian based operating system. I talked to the manjaro linux maintainers, they said that the next version of kodi will use the libcec version that comes with kodi.

Hi marshevms,
I am not using kodi and most probably I will not use it. I run raspbian and I was trying to start playing with CEC-frames just to power-on the tv as soon as I"airplay" something to my raspberry.

Now I have autodetect failed in all comands and -l gives me no adapter ...

satmandu, 2020-09-21:
@alecsandes FYI Raspbian is a debian based distribution...

escoand, 2021-02-21:
I'm experiencing the same problem running docker images on balenaos:

```
# balena run --rm -it --privileged balenalib/raspberrypi3-debian sh -c "install_packages cec-utils && cec-client -l"
...
libCEC version: 4.0.4, compiled on Linux-4.15.0-48-generic ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, AOCEC
Found devices: 1

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi

# balena run --rm -it --privileged balenalib/raspberrypi3-debian:bullseye sh -c "install_packages cec-utils && cec-client -l"
...
libCEC version: 6.0.2, compiled on Linux-4.19.0-12-armmp-lpae ... , features: P8_USB, DRM, P8_detect, randr, Exynos, AOCEC
Found devices: NONE
```

closed 2025-02-27 by github-actions[bot]