Pulse-Eight/libcec #531: cec-client generates segmentation fault after command seems to work?

Nodi-Rubrum, 2020-09-11:
Just recently cec-client generates segmentation fault after command seems to work.  This is a random issue, it will happen, then stop for many commands, then reappear on every command, until it disappears again.

The command sequence is almost always accept by the target device, and device executes said command, say pow, or on, or standby for example.  But cec-client monitor log returns sometimes 'unhandled response received: opcode=36 initiator=1 destination=0 response=0'  when this happens, the result code is 139, and cec-client failed with a segmentation fault error reported to console.

Details...
Raspberry Pi OS, latest image 05-27-2020 with all updates to date

\# uname -a
Linux eccentric.dachshund-digital.org 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
\# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

\# cec-client -i
libCEC version: 4.0.4, compiled on Linux-4.15.0-48-generic ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, AOCEC

Nodi-Rubrum, 2020-09-11:
A bit more information, when the seemingly random segmentation fault happens, in the testing I have done, most of the time, it is when a 'on' command is issued.  Some times the on command seems to complete, other times if clearly fails to complete before the segmentation fault is reported.  No sure if this is just because the fault reporting lags, due to code design or such.  Below is an example where issued on command and the segmentation fault was immediate, the receiving entity seemed to never get the cec command.

`Command failed:  /bin/echo on 0 | /usr/bin/cec-client -s -d 1
/bin/bash: line 1: 13627 Done
 /bin/echo on 0     13628 Segmentation fault      | /usr/bin/cec-client -s -d 1`

IMHO it almost looks like the cec-client (or library call?) chokes on the input stream/pipe content in this case.  But the on command is basic to any cec protocol use.

RichiG88, 2021-04-19:
I get the same error, when I use the command „echo tx 15:70:24:00 | /usr/bin/cec-client -s -d 1“ to activate the audio output on my av receiver. But I don’t get the error when I use „echo tx 1f:82:24:00 | /usr/bin/cec-client -s -d 1“. Both commands work nonetheless. I‘m running this on a Raspberry Pi 3B with Kernel 5.10.17-v7+ #1403

Jibun-no-Kage, 2025-02-26:
Cannot test with CEC 7.0... not available...
```
Get:1 http://deb.debian.org/debian bookworm/main arm64 libp8-platform2 arm64 2.1.0.1+dfsg1-4 [20.9 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libcec6 arm64 6.0.2-5+rpt2 [188 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 cec-utils arm64 6.0.2-5+rpt2 [30.0 kB]
```

Jibun-no-Kage, 2025-02-26:
Seems that original scan error fixed overall with Debian 12 based PiOS release?
```
# echo 'scan' | cec-client -d 1
log level set to 1
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect:
 path:     /sys/devices/platform/soc/3f902000.hdmi/cec0
 com port: /dev/cec0

opening a connection to the CEC adapter...
waiting for input
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```
But the active source issue remains.

Jibun-no-Kage, 2025-02-27:
I compiled and tested the latest release, i.e. 7.0 code base.  It fails the cec-client cannot auto detect any connected device on the cec bus.  The same hardware that worked above... fails with the latest code released on github.  Using Pi OS based on Debian 12, using a Pi Model 3 B, connected directly to a LG TV via HDMI port.

Under the 7.0 code... the device /dev/cec0 exists, but the cec-client fails to see it?  Even when I qualify the 'RPI' command option.  I followed the instructions for the on Pi device compile, no errors or exceptions via the make phases.  Opened a new issue show the test results I experienced.

Jibun-no-Kage, 2025-02-28:
No segment fault in the case of 7.0... the code just fails... it never discovers any applicable device, i,e. fails to see /dev/cec* devices.  Each issue represents a DIFFERENT and UNIQUE issue.  That needs to be addressed.

opdenkamp, 2025-02-28:
this was a ticket about a segmentation fault and that has been fixed. the problem that you've got is completely different, as I already answered in another ticket in which you've copy+pasted the same things. your problem is caused by your own build environment not recognising the Pi and that isn't a problem with libCEC

Jibun-no-Kage, 2025-02-28:
Then your instructions could be enhanced?  I followed them to the letter, and the result was not as presented.  I did not cross compile, I copied the commands directly from the 'guide' no errors or issues during cmake, make, make install... the result was no devices found even though /dev/cec0 exists.  There was not explicit explanation or notation of how to check cmake for RPI qualifier.