Pulse-Eight/libcec #533: [REQUEST] LibCEC support for new inexpensive Nvidia Jetson Nano single board computers?

Hedda, 2020-10-07:
Can we please get upstream libcec support for the new Nvidia Jetson Nano 2GB and Nvidia Jetson Nano 4GB budget SBC?

Nvidia Jetson Nano is based on a compute module with an SoC type similar to the Nvidia Tegra inside an Nvidia Shield TV box.

Not sure if you heard but Nvidia has released a budget series of Nvidia Jetson development kits called "Nvidia Jetson Nano".

Nvidia Jetson Nano 2GB was just launched and only cost $54 US

https://developer.nvidia.com/embedded/jetson-nano-2gb-developer-kit

Nvidia Jetson Nano 4GB has a slightly more powerful CPU and GPU plus extra expansion ports and still only cost $99 US

https://developer.nvidia.com/embedded/jetson-nano-developer-kit

There should make for more attractive HTPC hardware if media player applications like Kodi had HDMI-CEC support for it.

Some work has been done for the much more expensive Nvidia Jetson TK1 DevKit as well as Tegra X1/X2 SoCs for mobiles:

https://github.com/BuzzBumbleBee/libcec

https://forums.developer.nvidia.com/t/wip-hdmi-cec-on-tegra-jetson/35265/6

There is also a CEC driver available for Nvidia Tegra that uses the Linux Kernel's generic CEC framework interface:

https://github.com/torvalds/linux/blob/master/drivers/media/cec/platform/tegra/tegra_cec.c

http://lkml.iu.edu/hypermail/linux/kernel/1711.1/06245.html

https://www.mail-archive.com/linux-media@vger.kernel.org/msg118714.html

Hedda, 2020-11-04:
Any news on LibCEC support for Nvidia Jetson / Nvidia Tegra platform?

anishsane, 2021-09-24:
+1.
The lack of CEC support on Jetson Nano is preventing me to use it as a media player. (Lack of CEC and LIRC support)

malard, 2021-09-24:
It's not magic, we need access to the CEC pin on the hdmi port

If it's not connected then it will be impossible

anishsane, 2021-09-24:
I read in a [forum](https://forums.developer.nvidia.com/t/cec-not-working/73714/3) that the hardware is all fine - the CEC pin on the HDMI port is indeed connected. Only the software (kernel) does not support it.

DeanIsMe, 2021-09-25:
The HDMI-CEC pin is connected according to Nvidia's documentation.
I downloaded the Jetson Nano Carrier Board hardware design files (which are available from Nvidia, part # P3449-B01, document 600-13449-0000-400) and can confirm that HDMI-CEC, pin 13 on the HDMI port is connected to some IO circuitry which then connects to pin 94 on the SODIMM connector, to the Jetson Nano Module.
The datasheet of the Jetson Nano Module (which is also available from Nvidia, document DA-09366-001_v1.0) then describes pin 94 as HDMI_CEC on p94. The description of this pin is:
> Consumer Electronics Control (CEC) one-wire serial
> bus.
> NVIDIA provides low level CEC APIs (read/write).
> These are not supported in earlier Android releases.
> For additional CEC support, 3rd party libraries need to
> be made available.

anishsane, 2023-06-15:
[Android (LineageOS) is available on Jetson Nano.](https://wiki.lineageos.org/devices/porg/variant2/)
I have verified that CEC works in that build. So, from the hardware POV, things are OK.

anishsane, 2023-06-18:
I have created https://github.com/Pulse-Eight/libcec/pull/636

closed 2025-02-27 by github-actions[bot]