Pulse-Eight/libcec #535: libCEC 4.0.7 shows wrong serverVersion and iFirmwareversion in Kodi v18

Portisch, 2020-10-27:
When using tag libCEC 4.0.5 the correct libCEC version get shown in kodi.log and also in Kodi input settings:
`2020-10-27 18:10:59.622 T:4963    DEBUG <general>: InitialiseFeature - using libCEC v4.0.5`
[image]

When using tag 4.0.7 the wrong version get shown:
[image]
Also in kodi.log it shows first:
`2020-10-27 18:15:54.641 T:5115    DEBUG <general>: InitialiseFeature - using libCEC v4.0.4`

Later in kodi.log the correct version is shown:
`2020-10-27 18:15:55.780 T:5125    DEBUG <general>: CecLogMessage - CEC client registered: libCEC version = 4.0.7, client version = 4.0.0, firmware version = 5, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0,  compiled on 2020-10-26 17:50:58 by portisch@ubuntu on Linux 5.4.0-52-generic (x86_64), features: P8_USB, DRM, P8_detect, AOCEC`

I guess some switch if v5/6 is wrong for v4 as libCEC got set to v5, then v6 and then back to v4.
Kodi gets the version information by this function I guess:
https://github.com/xbmc/xbmc/blob/Leia/xbmc/peripherals/devices/PeripheralCecAdapter.cpp#L223

gibxxi, 2021-12-11:
The LibCEC.dll file shown in Kodi, is part of the Kodi distribution, and not provided by Pulse-Eight's driver/firmware package, *unless* at install time you checked the option in the CEC driver installer to enable Kodi Integration (which doesn't appear in all versions of the driver). Hence this is a Kodi issue, and not a fault of the LibCEC 4.0.7 driver package.

Also, all the 'Enable Kodi Integration' option does, is place a version of the 'libCEC.dll' driver, from that package, into the Kodi install directory, thus replacing the one provided by Kodi's own installer. The trouble with this method is, that the libCEC installer does not do a check of the Windows Registry to find out exactly *where* Kodi is installed to, and instead just dumps it into 'C:\Program Files\Kodi' (or C:\Program Files (x86)\Kodi - I forget which one it defaults to), so if you have Kodi  in an alternative folder, or on a different drive, the 'integration' won't work anyway.

Finally, I've tried copying the LibCEC.dll file provided by the installer package manually into the Kodi install folder (because of the above issue) and it simply causes Kodi to insta-CTD on launch anyway, so you aren't going to (probably) get much joy anyway.

It seems to me that with all the new hardware for 'Audio Professionals' Pulse-Eight have been pushing out lately, they've pretty much abandoned this device, at least for the time-being. None of the recent versions work with Kodi 18-19, and it seems like nobody cares. Which is disappointing considering we paid cash money for this thing, and it simply doesn't work as advertised.

Either way, your issue with incorrect version number being used in Kodi, is down to the version of the DLL bundled with (and installed by) Kodi itself, and not the software driver package available from Pulse-Eight, as the two are separate things, and the driver (without integration) is only utilized outside of Kodi / when Kodi is NOT running.

Dan / Gib.

malard, 2021-12-11:
Hi,

We were not aware that the integration had broken and can address this, we have not abandoned this device in any way, however we are like many manufacturers suffering from extreme component shortages so a lot of our global engineering teams time is spent working round supply constraints right now

I will make sure someone looks into this matter promptly

Thanks

Malard

Portisch, 2021-12-11:
Sorry, a year later a response?

We already using v6 for Kodi Matrix & Nexus.
We do not use the provided dll as we are Linux based and do compile the shared library from source. So it is a defect in the sources of libCEC.

Anyway I close this issue as this version is not active used anymore anyway.

closed 2021-12-11 by Portisch

gibxxi, 2021-12-11:
No offence intended here, This issue might have been posted a year ago - I saw it TODAY, and since I've also been haaving issues with the adapter, decided to intone.

I've seen that v6 is not supposed to be used with Kodi 18, but there's no indication of whether this ALSO holds true for Kodi 19. An udpate to the GitHUB wiki (clarifying this) would be welcome.

Also an actual readme or some form of documentation on what each setting in the Kodi (peripherals) settings area actually does, might go some way to me determining what (if anything) I may have configured wrong. With the best will in the world @malard some of the wording for the settings, at best, are borderline 'Engrish'.

Dan / Gib.

malard, 2021-12-11:
The main guy who wrote it is Dutch 😊, we can get that fixed also

gibxxi, 2021-12-12:
@malard

Then I thank you for your efforts and responses on this.

:)

Dan / Gib.

opdenkamp, 2021-12-13:
both libCEC 4.0 and Kodi v18 are obsolete. the version string issue has been fixed in libCEC 6 already.

as for the strings used in the add-on in Kodi: I did write the original ones, but they've been changed many times now by other Kodi devs...

Portisch, 2021-12-13:
> both libCEC 4.0 and Kodi v18 are obsolete

CoreELEC knows and use libCEC v6.
But XBMC do not know this fact:
https://github.com/xbmc/xbmc/blob/master/cmake/modules/FindCEC.cmake#L33
https://github.com/LibreELEC/LibreELEC.tv/pull/4755

gibxxi, 2021-12-13:
@opdenkamp

I've tried overwriting the cec.dll bundled with Kodi with the one bundled in the v6.0.2 package, but it just causes Kodi to insta-CTD.

The 32bit one fails with an error dialog, probably because Kodi is now a 64bit app, no? So either way, some investigation is required. Nothing in the logs that indicates an error. It gets as far as WASAPI device detection (and connection to the CEC-adapter) then just crashes.

Dan / Gib.

opdenkamp, 2021-12-13:
kodi has to be rebuilt from source if it got built using libcec v4. v4 is not binary compatible with v6, so you can't just copy + paste the new dll in without building kodi. so this is expected if kodi didn't pull the new version yet.

gibxxi, 2021-12-13:
Got it. Well if that's the case, I doubt very much they'll want to be upgrading to the later version via a point release, so it'll probably have to be Kodi 20. I'll try opening up an issue on the Kodi github - issues page about this instead. Thanks for the info.

Dan / Gib.

opdenkamp, 2021-12-13:
other than the wrong version info, 4.0.7 should work fine
it was a release that I built using the v4 api, but with some v6 fixes included. that allows you to use it in kodi when you just copy the dll in

gibxxi, 2021-12-13:
Unfortunately I don't thinik that works either. I'll try again, but the 4.0.7 edition I think also force closed Kodi last time I tried. Also, the cec-tray app is completely missing from the 4.0.7 zip package, any chance of adding that and pushing it out again?

opdenkamp, 2021-12-13:
if you need cec-tray, then you'll have to use v6, because it uses features that got added in v6
v4.0.7 was only built as backward compatibility release for kodi, which worked when I built it.

I'll get a new, proper, release out asap, but that may take some time with everything that's going on right now

gibxxi, 2021-12-13:
Aye, that's what I have been doing. Fortunately the files can be opened without installing with 7zip, which makes things a bit easier. Thanks for the help Lars.

:)

Dan / Gib.