Pulse-Eight/libcec #538: Second HDMI port on Raspberry Pi is now accessible to CEC.

audas, 2020-11-10:
I saw you acknowledged this in a previous comment - I have tested this with the new kernel and we are not able to work with the second port. Engineers at Raspi have suggested this is something that libcec will have to address - is there any chance of this being implemented, and if so any time soon ?

popcornmix, 2020-11-10:
To be more accurate, this is using the kms driver with support for /dev/cec0 and /dev/cec1.
Currently libCEC is hard coded to use /dev/cec0.

audas, 2020-11-10:
Thanks popcornmix I am very new and struggling and do not want to take up peoples time so any help is HUGE. I have narrowed things right down and have worked my way through a very steep learning curve  - so thank you again. 

I have set dtoverlay=vc4-kms-v3d in /boot/config.txt - no option shows up for it in the raspi-config and I think it is meant to on 5.9 kernel with latest Buster but not sure about that? I test run chrome and check GPU settings with massive 3d canvas tests and it works ok - but not sure of any flags to see if it is set or not. Another suggestion from 6by9 is to set video= in the cmdline.txt line in boot but I have no idea  "video="   what ? or just empty ? He doesn't say.

I am thinking of using your suggestion from another thread - https://github.com/raspberrypi/firmware/issues/1342

You suggest installing it twice with /dev/cec0 and /dev/cec1 hard coded. Do you know where these are located ?
Do I download the changed code and then install it from a local directory on pi ? I don't want to create a new git (sorry a bit new).

There is the possibility of setting hdmi_preferred_port=1 in the config.txt and rebooting and doing things and then rebooting again - really would prefer a software fix though and have no idea if this would even work. Next test.

Also you suggest the following - does this mean I include it with the other cmake flags

cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. && \
make -j4

Do I add it to the existing flags

cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..

So it looks like this

cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. && \

And last question - you say Firmware does not matter - however someone on this thread suggested it be upgraded - I have no idea how that would happen. Mine says =1

But they also say this

**dtoverlay=vc4-kms-v3d-pi4 should not be necessary with an up-to-date firmware - dtoverlay=vc4-kms-v3d**

Is that -pi4 at the end of the driver overlay name something I should be using ?

https://github.com/raspberrypi/linux/issues/3701

"You are my last hope."

popcornmix, 2020-11-10:
As I said elsewhere, build with:
```
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. && \
make -j4
```
You don't need any include directories with building for the linux api.

Check if cec-client work on hdmi0. If it does you can edit 
https://github.com/Pulse-Eight/libcec/blob/76551ea1dd9a55f0ce1533e440dc12dbc594f7ba/include/cectypes.h#L287

to /dev/cec1 and rebuild and see if it works for hdmi1.

If that works, you could stick with two binaries (plus libs) but it would be nicer to make that switchable with a command line option.

audas, 2020-11-12:
Thanks popcornmix - cec-ctl seems to be working much better than cec-client for now on the new Raspberry Kernel - works with both ports so this is the way forward I think.

Doesn't look like there is much interest in addressing the issues on this as the  situation was raised several months ago and nothing has been done and it really does seem like a trivial fix.

Having two binaries just appears very messy unfortunately.

Kwiboo, 2020-11-12:
If possible please test https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch, that patch/commit is intended to add support for multiple `/dev/cecX` devices, I have however only managed to tested it on devices with a single `/dev/cec0` device with success so far.

I will create a pull request as soon as I get confirmation it enables support for multiple `/dev/cecX` devices, so that the implementation details can be discussed (inspired from usb adapter detection).

audas, 2020-11-12:
> If possible please test https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch, that patch/commit is intended to add support for multiple `/dev/cecX` devices, I [...]

I have multiple devices and lots of televisions and can test anything. I am not skilled enough to know how to implement this patch though. I use the following to implement the library - I delete the existing directories and just repeat. How do I go about including this patch ?

>sudo apt-get update
sudo apt-get install cmake libudev-dev libxrandr-dev python3-dev swig git
cd
git clone https://github.com/Pulse-Eight/platform.git
mkdir platform/build
cd platform/build
cmake ..
make
sudo make install
cd
git clone https://github.com/Pulse-Eight/libcec.git
mkdir libcec/build
cd libcec/build
cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
make -j4
sudo make install
sudo ldconfig

popcornmix, 2020-11-12:
Try: 
```
curl -L https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch | git am
```

audas, 2020-11-12:
Do I have to rebuild the binaries with your hardware adjustment ?

So the following from terminal ?

>curl -L https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch | git am
cd libcec/build
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. && \
make -j4
sudo make install
sudo ldconfig

popcornmix, 2020-11-12:
Yes, that looks about right.
You are patching the git tree, so in your original list it would be entered after the git clone (from within the git directory tree).

audas, 2020-11-12:
ok so remove directory and its contents, git clone, patch, create directory, cmake with your changes ?

Thanks for everyones help. I'll post back in an hour.

>rm -r libcec/build
git clone https://github.com/Pulse-Eight/libcec.git
curl -L https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch | git am
mkdir libcec/build
cd libcec/build
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. && \
make -j4
sudo make install
sudo ldconfig

audas, 2020-11-12:
>curl -L https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch | git am

curl: (23) Failed to writing body (937 != 1370)

I checked diskspace and used sudo - but it fails.

popcornmix, 2020-11-12:
Doesn't do that here. What about:
```
wget https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch
git am master...Kwiboo:cec-framework-multiple.patch
```

audas, 2020-11-12:
i removed the -L (--location ?) and then just used 

>curl https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch | git apply -

This threw multiple errors about file locations. I am testing this on raspberry and that file looks to be dealing with linux specifically?

So errors were

>error: include/cectypes.h: No such file or directory
error: src/libcec/adapter/AdapterFactory.cpp: No such file or directory
error: src/libcec/adapter/Linux/LinuxCECAdapterCommunication.cpp: No such file or directory
error: src/libcec/adapter/Linux/LinuxCECAdapterCommunication.h: No such file or directory
error: src/libcec/adapter/Llinux/LinuxCECAdapterDetection.cpp: No such file or directory
error: src/libcec/adapter/Llinux/LinuxCECAdapterDetection.h: No such file or directory

popcornmix, 2020-11-12:
git am not git apply. And were you in libcec directory?

audas, 2020-11-12:
I was not in the libcec directory.
Tried it from there as well and it gave me the same error.

I ran the wget from from the directory and it is throwing a git error regarding not having my email address ?

>*** please tell me who you are
Run

git config --global user.email "you@example.com"
git config --global user.name "Your Name"

to set your accoutns etc

fatal: unable to auto-detect email address (got 'pi@taspberrypi.(none)'

popcornmix, 2020-11-12:
Running:
```
git config --global user.email "you@example.com"
git config --global user.name "Your Name"
```
should work (or you could your real name/email)

audas, 2020-11-12:
seemed to work with the wget.

I applied the kernel cmake command

>cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0 .. &&

and it just hung the terminal. Nothing happens - it just shows a single > symbol and thats it. I have not seen that before?

audas, 2020-11-12:
Ok - removed the double ampersand and it compiled. 

Ports are not switching. Which is really strange as previously - it would switch, now it will only try and access first port.
It says 
>"using HDMI port 2"

But then it says

>"no serial port given. trying to autodetect : 
path: /sys/devices/platofrom/soc/fef00700.hdmi/cec0
com port: /dev/cec0

So it is not passing through the fact it is switching ports to the new patch - which is then reverting to look for one.

I have tried with everything -p --port 0,1,2 /dev/cec1  etc

rijotech, 2022-07-04:
The path for multiple cec dev does not work.

popcornmix, 2022-07-04:
> The path for multiple cec dev does not work.

You'll have to be more specific. The legacy cec interface (driven by firmware) only supports HDMI0.
The newer `/dev/cec0`/`/dev/cec1` kernel driver does support both hdmi ports.
The libcec included in RpiOS bullseye should support this.

inspector71, 2024-02-21:
> > The path for multiple cec dev does not work.
> 
> You'll have to be more specific. The legacy CEC interface (driven by firmware) only supports `HDMI0`. The newer `/dev/cec0` and `/dev/cec1` kernel driver does support both HDMI ports. The `libcec` included in RPiOS `bullseye` should support this.

Where you say "libcec included in RpiOS bullseye", could you please be more specific? Are you referring to a custom `libcec` build or a direct build from the `libcec` repository, without any modifications? 

What is the minimum `libcec` version, as bundled with RPiOS, that supports both HDMI ports, via `/dev/cec0` and `/dev/cec1` ?

By "support this", are you referring to _concurrent support_ of _both_ interfaces? Or are you referring to enabling `CEC` on _either interface_, but **not** _concurrently_?

Thank you most appreciatively for your valued time.

closed 2025-02-27 by github-actions[bot]