Pulse-Eight/libcec #540: ubuntu on Raspberry PI 4: autodetect failed

mihaiav, 2020-11-18:
I have Ubuntu  installed on a Raspberry PI4 and it seems CEC fails to work. The error is:
```
$sudo cec-client
No device type given. Using 'recoding device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect: FAILED
```

uname

```
Linux 5.8.0-1007-raspi \#10-Ubuntu SMP PREEMPT aarch64 aarch64 aarch64 GNU/Linux
```

nsaeger1, 2020-11-19:
I am having auto detect failed as well on rasp pi 4 running ubuntu 20.10

`$ echo 'scan' | cec-client -s -d 1
autodetect FAILED
$ sudo cec-client -i
libCEC version: 6.0.2, compiled on Linux-4.15.0-109-generic ... , features: P8_USB, DRM, P8_detect, randr, Exynos, AOCEC
$ uname -a
Linux desktop 5.8.0-1006-raspi #9-Ubuntu SMP PREEMPT Fri Oct 16 12:55:30 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.10
Release:        20.10
Codename:       groovy`

graysky2, 2020-11-28:
I can also confirm seeing a failure on 64-bit install of Arch ARM.  Latest kernel from rpi-5.10.y branch/firmware.
```
% echo "scan" | cec-client RPI -s -d 1
opening a connection to the CEC adapter...
ERROR:   [              11]   Open - vc_cec could not be initialised
ERROR:   [              11]   could not open a connection (try 1)
ERROR:   [            1011]   Open - vc_cec could not be initialised
ERROR:   [            1011]   could not open a connection (try 2)
ERROR:   [            2012]   Open - vc_cec could not be initialised
ERROR:   [            2012]   could not open a connection (try 3)
ERROR:   [            3012]   Open - vc_cec could not be initialised
ERROR:   [            3012]   could not open a connection (try 4)
ERROR:   [            4013]   Open - vc_cec could not be initialised
ERROR:   [            4013]   could not open a connection (try 5)
ERROR:   [            5013]   Open - vc_cec could not be initialised
ERROR:   [            5013]   could not open a connection (try 6)
ERROR:   [            6014]   Open - vc_cec could not be initialised
ERROR:   [            6014]   could not open a connection (try 7)
ERROR:   [            7014]   Open - vc_cec could not be initialised
ERROR:   [            7014]   could not open a connection (try 8)
ERROR:   [            8015]   Open - vc_cec could not be initialised
ERROR:   [            8015]   could not open a connection (try 9)
ERROR:   [            9015]   Open - vc_cec could not be initialised
ERROR:   [            9015]   could not open a connection (try 10)
unable to open the device on port RPI
ERROR:   [           10016]   could not start CEC communications
```

This is in contrast to the 32-bit version of the Distro, armv7h, again with the latest kernel from rpi-5.10.y:
```
% echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```

@opdenkamp - Any thoughts?
@mihaiav - Please edit the title of your issue to reflect that you're using 64-bit distro.

graysky2, 2020-11-29:
I tried running the command on aarch64 a few more times and got it to work but it ended in a seg fault.  I do not know why it worked and I have been unable to get it to work since this :confused: 

```
% echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
[1]    288 done                              echo 'scan' | 
       289 segmentation fault (core dumped)  cec-client -s -d 1

# coredumpctl info
           PID: 751 (cec-client)
           UID: 1000 (facade)
           GID: 985 (users)
        Signal: 11 (SEGV)
     Timestamp: Sun 2020-11-29 05:59:59 EST (39s ago)
  Command Line: cec-client -s -d 1
    Executable: /usr/bin/cec-client-6.0.2
 Control Group: /user.slice/user-1000.slice/session-1.scope
          Unit: session-1.scope
         Slice: user-1000.slice
       Session: 1
     Owner UID: 1000 (facade)
       Boot ID: a87639a315444ab58d4ae48cdef7d40d
    Machine ID: ef6412950137417eb861dad2ace857f9
      Hostname: workbench
       Storage: /var/lib/systemd/coredump/core.cec-client.1000.a87639a315444ab58d4ae48cdef7d40d.751.1606647599000000.zst
       Message: Process 751 (cec-client) of user 1000 dumped core.
                
                Stack trace of thread 756:
                #0  0x0000007f8d679e48 _ZN3CEC26CRPiCECAdapterMessageQueue15MessageReceivedENS_10cec_opcodeENS_19cec_logical_addressES2_j (libcec.so.6 + 0x58e48)
                #1  0x0000007f8d40dc04 cecservice_notify_func (libbcm_host.so + 0x10c04)
                #2  0x0000007f8d44a760 vcos_thread_entry (libvcos.so + 0x3760)
                #3  0x0000007f8dbf2f44 start_thread (libpthread.so.0 + 0x7f44)
                #4  0x0000007f8d7c695c thread_start (libc.so.6 + 0xd395c)
                
                Stack trace of thread 751:
                #0  0x0000007f8dbf974c pthread_cond_wait@@GLIBC_2.17 (libpthread.so.0 + 0xe74c)
                #1  0x0000007f8d678010 _ZN3CEC27CRPiCECAdapterCommunication24UnregisterLogicalAddressEv (libcec.so.6 + 0x57010)
                #2  0x0000007f8d6782c8 _ZN3CEC27CRPiCECAdapterCommunicationD1Ev (libcec.so.6 + 0x572c8)
                #3  0x0000007f8d67852c _ZN3CEC27CRPiCECAdapterCommunicationD0Ev (libcec.so.6 + 0x5752c)
                #4  0x0000007f8d666990 _ZN3CEC13CCECProcessor5CloseEv (libcec.so.6 + 0x45990)
                #5  0x000000555e0f2f48 n/a (cec-client-6.0.2 + 0x2f48)
                #6  0x0000007f8d717538 __libc_start_main (libc.so.6 + 0x24538)
                #7  0x000000555e0f3c98 n/a (cec-client-6.0.2 + 0x3c98)
                #8  0x000000555e0f3c98 n/a (cec-client-6.0.2 + 0x3c98)
```

JordanDDisch, 2021-01-20:
if you don't need ubuntu you can 'sudo apt install cec-utils' on rasbian and it will work.

HeadFox, 2021-12-25:
I tried to follow many guide but this one was the one to make it work on Debian 11 : https://github.com/Pulse-Eight/libcec/issues/470#issuecomment-629702541

closed 2025-02-27 by github-actions[bot]