Pulse-Eight/libcec #542: Showing all data sent by TV

mikouaji, 2020-11-26:
Hello,
my problem is that I don't see any activity from my TV remote control in cec-client.
I can send signals from the RPI to the TV such like turning it on or switching HDMI ports, but I dont see any remote buttons pressed activity in cec-client. 
Does that mean that my TV doesnt send them via cec?
Is there an option to see all the data sent between the RPI and the TV or all signals sent by the TV?
I'm using cec on a RPI4 newest RaspberryPI OS based on Debian Buster. 
The [TV](https://www.orbitadigital.com/en/discontinued/1608-tv-ferguson-t232fhd506-32-fullhd-1080p-dvb-t2.html) is some random Chinese TV rebranded to be sold in EU, I dont have any other right now to test this somewhere else.
If there is anymore info I can provide just ask :)

malard, 2020-11-26:
if your TV doesnt support CEC then we wont see anything, if you can't even state the brand name and/or model number of your TV then I am unable to help

closed 2020-11-26 by malard

mikouaji, 2020-11-26:
The tv does support CEC, I have it enabled in the settings and as stated in the issue I can send signals such as switching input sources or turning it on from the RPI.
The problem is with seeing remote presses
The tv model was stated under the url, its TV Ferguson T232FHD506
Here is the manual http://ferguson-digital.eu/?wpfb_dl=1183 I didnt find any help in it.

malard, 2020-11-26:
run cec-client in monitoring mode, type cec-client --help

mikouaji, 2020-11-26:
at line 16 I've run the scan command
```
pi@raspberrypi:~ $ cec-client -m
starting a monitor-only client. use 'mon 0' to switch to normal mode
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect: 
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [              68]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              68]	Open - vc_cec initialised
DEBUG:   [              68]	logical address changed to Free use (e)
NOTICE:  [              68]	connection opened
waiting for input
DEBUG:   [              69]	processor thread started
scan
requesting CEC bus information ...
DEBUG:   [           26332]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [           26332]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26332]	<< e0
DEBUG:   [           26362]	>> POLL sent
DEBUG:   [           26362]	TV (0): device status changed into 'present'
DEBUG:   [           26363]	<< Broadcast (F) -> Recorder 1 (1): POLL
DEBUG:   [           26363]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26363]	<< e1
DEBUG:   [           26453]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26453]	<< e1
DEBUG:   [           26547]	>> POLL not sent
DEBUG:   [           26547]	Recorder 1 (1): device status changed into 'not present'
DEBUG:   [           26547]	<< Broadcast (F) -> Recorder 2 (2): POLL
DEBUG:   [           26547]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26547]	<< e2
DEBUG:   [           26637]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26637]	<< e2
DEBUG:   [           26728]	>> POLL not sent
DEBUG:   [           26728]	Recorder 2 (2): device status changed into 'not present'
DEBUG:   [           26728]	<< Broadcast (F) -> Tuner 1 (3): POLL
DEBUG:   [           26728]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26728]	<< e3
DEBUG:   [           26818]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26818]	<< e3
DEBUG:   [           26908]	>> POLL not sent
DEBUG:   [           26908]	Tuner 1 (3): device status changed into 'not present'
DEBUG:   [           26908]	<< Broadcast (F) -> Playback 1 (4): POLL
DEBUG:   [           26908]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           26908]	<< e4
DEBUG:   [           27000]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           27000]	<< e4
DEBUG:   [           27090]	>> POLL not sent
DEBUG:   [           27090]	Playback 1 (4): device status changed into 'not present'
DEBUG:   [           27090]	<< Broadcast (F) -> Audio (5): POLL
DEBUG:   [           27090]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           27091]	<< e5
DEBUG:   [           27181]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           27181]	<< e5
DEBUG:   [           27271]	>> POLL not sent
DEBUG:   [           27271]	Audio (5): device status changed into 'not present'
DEBUG:   [           27271]	<< Broadcast (F) -> Tuner 2 (6): POLL
DEBUG:   [           27271]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           27271]	<< e6
DEBUG:   [           27362]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           27362]	<< e6
DEBUG:   [           27452]	>> POLL not sent
DEBUG:   [           27452]	Tuner 2 (6): device status changed into 'not present'
DEBUG:   [           27452]	<< Broadcast (F) -> Tuner 3 (7): POLL
DEBUG:   [           27452]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           27452]	<< e7
DEBUG:   [           27542]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
[... 68 lines cut ...]
DEBUG:   [           29616]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           29616]	<< e0:8f
TRAFFIC: [           29772]	>> 0e:90:00
DEBUG:   [           29772]	TV (0): power status changed from 'unknown' to 'on'
DEBUG:   [           29772]	expected response received (90: report power status)
DEBUG:   [           29772]	<< requesting menu language of 'TV' (0)
DEBUG:   [           29772]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           29772]	<< e0:91
TRAFFIC: [           29976]	>> 0f:32:70:6f:6c
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   1.4
power status:  on
language:      pol

currently active source: unknown (-1)
DEBUG:   [           29976]	TV (0): menu language set to 'pol'
DEBUG:   [           29976]	expected response received (32: set menu language)
DEBUG:   [           29976]	>> TV (0) -> Broadcast (F): set menu language (32)
```