Pulse-Eight/libcec #543: Problems using cec-client on 64-bit installation on RPi4

graysky2, 2020-11-29:
Similar to https://github.com/Pulse-Eight/libcec/issues/540, I get failures on a 64-bit install of Arch ARM on RPi4.  Latest kernel from rpi-5.10.y branch/firmware.
```
% echo "scan" | cec-client RPI -s -d 1
opening a connection to the CEC adapter...
ERROR:   [              11]   Open - vc_cec could not be initialised
ERROR:   [              11]   could not open a connection (try 1)
ERROR:   [            1011]   Open - vc_cec could not be initialised
ERROR:   [            1011]   could not open a connection (try 2)
ERROR:   [            2012]   Open - vc_cec could not be initialised
ERROR:   [            2012]   could not open a connection (try 3)
ERROR:   [            3012]   Open - vc_cec could not be initialised
ERROR:   [            3012]   could not open a connection (try 4)
ERROR:   [            4013]   Open - vc_cec could not be initialised
ERROR:   [            4013]   could not open a connection (try 5)
ERROR:   [            5013]   Open - vc_cec could not be initialised
ERROR:   [            5013]   could not open a connection (try 6)
ERROR:   [            6014]   Open - vc_cec could not be initialised
ERROR:   [            6014]   could not open a connection (try 7)
ERROR:   [            7014]   Open - vc_cec could not be initialised
ERROR:   [            7014]   could not open a connection (try 8)
ERROR:   [            8015]   Open - vc_cec could not be initialised
ERROR:   [            8015]   could not open a connection (try 9)
ERROR:   [            9015]   Open - vc_cec could not be initialised
ERROR:   [            9015]   could not open a connection (try 10)
unable to open the device on port RPI
ERROR:   [           10016]   could not start CEC communications
```

This is in contrast to the 32-bit version of the Distro, armv7h, again with the latest kernel from rpi-5.10.y:
```
% echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```

I tried several times and usually get the above on the 64-bit install.  I found that if I let the box boot and sit for a while, I am able to get `cec-client` to connect but doing so ends in a core dump.

```
% echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
[1]    288 done                              echo 'scan' | 
       289 segmentation fault (core dumped)  cec-client -s -d 1
```

Here is the info on the core dump.
```
% coredumpctl dump --output=dump
           PID: 346 (cec-client)
           UID: 1000 (facade)
           GID: 985 (users)
        Signal: 11 (SEGV)
     Timestamp: Sun 2020-11-29 07:31:32 EST (50s ago)
  Command Line: cec-client -s -d 1
    Executable: /usr/bin/cec-client-6.0.2
 Control Group: /user.slice/user-1000.slice/session-3.scope
          Unit: session-3.scope
         Slice: user-1000.slice
       Session: 3
     Owner UID: 1000 (facade)
       Boot ID: f037c05e68814ecf8112246aa79b4c18
    Machine ID: ef6412950137417eb861dad2ace857f9
      Hostname: workbench
       Storage: /var/lib/systemd/coredump/core.cec-client.1000.f037c05e68814ecf8112246aa79b4c18.346.1606653092000000.zst
       Message: Process 346 (cec-client) of user 1000 dumped core.
                
                Stack trace of thread 353:
                #0  0x0000007f86eb1e48 _ZN3CEC26CRPiCECAdapterMessageQueue15MessageReceivedENS_10cec_opcodeENS_19cec_logical_addressES2_j (libcec.so.6 + 0x58e48)
                #1  0x0000007f86c45c04 cecservice_notify_func (libbcm_host.so + 0x10c04)
                #2  0x0000007f86c82760 vcos_thread_entry (libvcos.so + 0x3760)
                #3  0x0000007f8742af44 start_thread (libpthread.so.0 + 0x7f44)
                #4  0x0000007f86ffe95c thread_start (libc.so.6 + 0xd395c)
                
                Stack trace of thread 351:
                #0  0x0000007f87434430 do_futex_wait.constprop.0 (libpthread.so.0 + 0x11430)
                #1  0x0000007f8743455c __new_sem_wait_slow.constprop.0 (libpthread.so.0 + 0x1155c)
                #2  0x0000007f86c49ff0 dispmanx_notify_func (libbcm_host.so + 0x14ff0)
                #3  0x0000007f86c82760 vcos_thread_entry (libvcos.so + 0x3760)
                #4  0x0000007f8742af44 start_thread (libpthread.so.0 + 0x7f44)
                #5  0x0000007f86ffe95c thread_start (libc.so.6 + 0xd395c)
                
                Stack trace of thread 350:
                #0  0x0000007f86ff6b50 ioctl (libc.so.6 + 0xcbb50)
                #1  0x0000007f86c685f0 completion_thread (libvchiq_arm.so + 0x25f0)
                #2  0x0000007f86c82760 vcos_thread_entry (libvcos.so + 0x3760)
                #3  0x0000007f8742af44 start_thread (libpthread.so.0 + 0x7f44)
                #4  0x0000007f86ffe95c thread_start (libc.so.6 + 0xd395c)
                
                Stack trace of thread 352:
                #0  0x0000007f87434430 do_futex_wait.constprop.0 (libpthread.so.0 + 0x11430)
                #1  0x0000007f8743455c __new_sem_wait_slow.constprop.0 (libpthread.so.0 + 0x1155c)
                #2  0x0000007f86c43648 tvservice_notify_func (libbcm_host.so + 0xe648)
                #3  0x0000007f86c82760 vcos_thread_entry (libvcos.so + 0x3760)
                #4  0x0000007f8742af44 start_thread (libpthread.so.0 + 0x7f44)
                #5  0x0000007f86ffe95c thread_start (libc.so.6 + 0xd395c)
                
                Stack trace of thread 346:
                #0  0x0000007f8743174c pthread_cond_wait@@GLIBC_2.17 (libpthread.so.0 + 0xe74c)
                #1  0x0000007f86eb0010 _ZN3CEC27CRPiCECAdapterCommunication24UnregisterLogicalAddressEv (libcec.so.6 + 0x57010)
                #2  0x0000007f86eb02c8 _ZN3CEC27CRPiCECAdapterCommunicationD1Ev (libcec.so.6 + 0x572c8)
                #3  0x0000007f86eb052c _ZN3CEC27CRPiCECAdapterCommunicationD0Ev (libcec.so.6 + 0x5752c)
                #4  0x0000007f86e9e990 _ZN3CEC13CCECProcessor5CloseEv (libcec.so.6 + 0x45990)
                #5  0x000000555c5d2f48 n/a (cec-client-6.0.2 + 0x2f48)
                #6  0x0000007f86f4f538 __libc_start_main (libc.so.6 + 0x24538)
                #7  0x000000555c5d3c98 n/a (cec-client-6.0.2 + 0x3c98)
                #8  0x000000555c5d3c98 n/a (cec-client-6.0.2 + 0x3c98)
```

I have a 58M coredump I can share if needed.

posuch, 2020-12-07:
confirming the beahviour as described 

see two screens bellow how the cec presents itself in Kodi - for 32bit arm the firmware version is 1 for 64bit the firmware version is 65535

not sure if this helps , just posting how it looks like ( more like a symptom )

32bit

[image]

64bit

[image]

henkv1, 2021-01-29:
I have the same issue on a 32 bit installation and full kms (vc4-kms-v3d):

# echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   unknown
power status:  unknown
language:      ???

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
Segmentation fault (core dumped)

It works fine when I use fake kms (vc4-fkms-v3d) on the same system:
# echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      dut

device #1: Recorder 1
address:       1.0.0.0
active source: yes
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: Recorder 1 (1)

graysky2, 2021-01-29:
@henkv1 - Wondering if this is more a function of kernel code rather than Pulse-Eight code given your observation.  Would you create a ticket against the [kernel](https://github.com/raspberrypi/linux/issues) and let's see what they say?

henkv1, 2021-01-29:
@graysky2 - I created a ticket for this: https://github.com/raspberrypi/linux/issues/4103

graysky2, 2021-01-29:
Thanks.  I found a solution/work-around in that issue.  Please let me know if it's working for you.

graysky2, 2021-01-29:
@opdenkamp - It seems the aarch64 was a red herring.  The issue could be that when we boot using `vc4-kms-v3d` driver, `cec-client` cannot connect but if we boot using `vc4-fkms-v3d` it does in fact connect.

How can we build it to allow for either of these?

* If I build with `-DHAVE_LINUX_API=1` it works with kms.
* If I build with both `-DHAVE_LINUX_API=1` and `-DHAVE_RPI_API=1` it does NOT work with kms.

popcornmix, 2021-02-01:
I've just tried this. I'm on
```
$ uname -a
Linux domnfs 5.10.11-v7l+ #245 SMP Fri 29 Jan 19:08:00 GMT 2021 armv7l GNU/Linux
```
which was a kernel build that includes https://github.com/raspberrypi/linux/pull/3999 (I don't think that will have fixed this)

I built master branch of libcec with:
```
mkdir -p build_both && cd build_both
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=1 -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib .. && \
make -j4
sudo make install && \
sudo ldconfig
```
```
pi@pi4:~ $ cec-client -i
libCEC version: 5.0.0, git revision: libcec-4.0.4+62-b7fd475, compiled on 2021-02-01 14:11:59 by pi@domnfs on Linux 5.10.11-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
```
I tested with (each one in turn):
```
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-kms-v3d
```
each time I ran cec-client and pressed buttons on remote and they logged messages. e.g.
```
DEBUG:   [           12571]	>> TV (0) -> Recorder 1 (1): user control pressed (44)
```
with kms:
```
pi@pi4:~ $ cec-client -l
libCEC version: 5.0.0, git revision: libcec-4.0.4+62-b7fd475, compiled on 2021-02-01 14:11:59 by pi@domnfs on Linux 5.10.11-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
ERROR:   [            3601]	error opening serial port '': No such file or directory
ERROR:   [            3601]	could not open a connection (try 1)
ERROR:   [            6438]	error opening serial port '': No such file or directory
ERROR:   [            6438]	could not open a connection (try 2)
ERROR:   [            8328]	error opening serial port '': No such file or directory
ERROR:   [            8328]	could not open a connection (try 3)
ERROR:   [            9588]	error opening serial port '': No such file or directory
ERROR:   [            9588]	could not open a connection (try 4)
Found devices: 3

device:              1
com port:            /dev/cec0
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

device:              2
com port:            /dev/cec1
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

device:              3
com port:            
vendor id:           0000
product id:          0000
firmware version:    65535
```
with fkms or none:
```
pi@pi4:~/projects/libcec $ cec-client -l
libCEC version: 5.0.0, git revision: libcec-4.0.4+62-b7fd475, compiled on 2021-02-01 14:11:59 by pi@domnfs on Linux 5.10.11-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
Found devices: 1

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi
```

ioogithub, 2021-03-22:
I have been trying to get my TV remote working with my Raspberry Pi 4 Model B and the 64 bit OS without success.

> Note that on aarch64, I can add `hdmi_force_hotplug=1` to `/boot/config.txt` and I can get `cec-client` to connect

I tried adding this hotplug option but it didn't make any difference:

```
$ cec-client -l
Found devices: NONE
```

> @opdenkamp - It seems the aarch64 was a red herring. The issue could be that when we boot using vc4-kms-v3d driver, > cec-client cannot connect but if we boot using vc4-fkms-v3d it does in fact connect.

> dtoverlay=vc4-kms-v3d
> dtoverlay=vc4-fkms-v3d
> #dtoverlay=vc4-kms-v3d

Ive tried setting all of these all of these and none of them worked.

> Thanks. I found a solution/work-around in that issue. Please let me know if it's working for you.

What solution did you ultimately find? I followed the linked thread but it was a bit over my head.

`uname -a`
Linux raspberrypi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb22 11:37 GMT 2021 aarch64 GNU/Linux

popcornmix, 2021-03-23:
If using the kms driver (`dtoverlay=vc4-kms-v3d`) you need libcec to be built with right options.
Try building this lib with:
```
cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0
```

ioogithub, 2021-03-23:
> If using the kms driver (`dtoverlay=vc4-kms-v3d`)
I am using the fkms which I believe is the default and this doesn't work. Is using the kms driver built with the right options the only way currently to get it working?

I have never build a linux driver before, is there a guide somewhere for doing this?

Do you think this will be fixed eventually?

graysky2, 2021-03-23:
@ioogithub - Driver or program... doesn't matter.  You just need the build dependencies and you build it.  For a recipe, see [this](https://github.com/archlinuxarm/PKGBUILDs/blob/34489de78c1a5124f85651a26862f46aeece4e06/alarm/libcec-rpi/PKGBUILD).

popcornmix, 2021-03-23:
Building instructions are here:
https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md

I don't know if there is anything to fix. This repo contains code that works with 64-bit kms kernel.

If you want a prebuilt library in the distribution of your choice, then you can request that there.

Jibun-no-Kage, 2022-04-11:
This issue was never closed? Was it resolved?  I have a Vizio TV (32in) that CEC and Pi have been working fine with, now I can seeing all kinds of CEC_TRANSMIT errors.  Pi 3B+ running 64 bit.  But I am not running KMS but FKMS driver.  A CEC scan works.  Get power state, power off and power on work.  But switching from HDMI1 to HDMI2 or back HDMI2 to HDMI1 fails via cec-client.  This was working fine before moving to 64 bit and Bullseye recently.  I have swapped cables, done a hard reset of the TV, even used a new Pi OS 11 image on a different SD card, but I still can't get cec-client to switch HDMI ports via CEC protocol.

inspector71, 2024-01-21:
> @opdenkamp - It seems the aarch64 was a red herring. The issue could be that when we boot using `vc4-kms-v3d` driver, `cec-client` cannot connect but if we boot using `vc4-fkms-v3d` it does in fact [...]

But, does it work with Fakey `fKMS` ?

graysky2, 2024-01-21:
https://github.com/Pulse-Eight/libcec/issues/543#issuecomment-805061916

closed 2024-01-21 by graysky2

jbumgit, 2024-11-08:
I'm on Pi 4 running Pi OS (bookworm, 32-bit).  I built libcec with the recommended steps using -DHAVE_LINUX_API=1.

```
~/libcec/build $ cmake -DHAVE_LINUX_API=1 ..
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython3.11.so (found version "3.11.2") 
-- git found: libcec-6.0.2+3-bf5a97d
CMake Deprecation Warning at src/libcec/cmake/CheckPlatformSupport.cmake:224 (cmake_policy):
  The OLD behavior for policy CMP0078 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  src/libcec/CMakeLists.txt:137 (include)

CMake Deprecation Warning at src/libcec/cmake/CheckPlatformSupport.cmake:228 (cmake_policy):
  The OLD behavior for policy CMP0086 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  src/libcec/CMakeLists.txt:137 (include)

-- Configured features:
-- Pulse-Eight CEC Adapter:                yes
-- Pulse-Eight CEC Adapter detection:      yes
-- xrandr support:                         yes
-- Raspberry Pi support:                   no
-- TDA995x support:                        no
-- Exynos support:                         no
-- DRM support:                            yes
-- Linux support:                          yes
-- AOCEC support:                          no
-- i.MX6 support:                          no
-- Python support:                         version 3.11.2 (3.11)
```

I'm running the "kms" driver that appears to be the default for bookworm.  I'm still not able to get my remote to control Kodi.  My Pi also does not wake up the TV when it boots up like previous Pis used to.

I have a micro HDMI to HDMI adapter... wondering if that's also a factor here?

My cec scan shows the expected cec devices under /dev.

```
~$ cec-client -l
libCEC version: 6.0.2, compiled on Linux ... , features: P8_USB, DRM, P8_detect, randr, Exynos, Linux, AOCEC
Found devices: 2

device:              1
com port:            /dev/cec0
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

device:              2
com port:            /dev/cec1
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

```
But it's not able to open a connection.  I have tried using recording/playback/tv for the device type.

```
~ $ cec-client --playback
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
opening a connection to the CEC adapter...
DEBUG:   [               1]	Broadcast (F): osd name set to 'Broadcast'
ERROR:   [            3504]	error opening serial port '--playback': No such file or directory
ERROR:   [            3504]	could not open a connection (try 1)
ERROR:   [            6339]	error opening serial port '--playback': No such file or directory
ERROR:   [            6339]	could not open a connection (try 2)
ERROR:   [            8229]	error opening serial port '--playback': No such file or directory
ERROR:   [            8229]	could not open a connection (try 3)
ERROR:   [            9489]	error opening serial port '--playback': No such file or directory

```

Any help would be appreciated.

jbumgit, 2024-11-10:
Well, at least for my case, it was the micro HDMI to HDMI adapter I was using for my Pi 4B.  It claimed to be able to do HDMI-CEC when I had purchased it over a year ago.  Anyway, I purchased an Amazon Basics micro HDMI-HDMI cable (not adapter) that I saw mentioned by various folks as working for their PI...  whadya know... that did it for me.