Pulse-Eight/libcec #546: Libcec tray app forcing computer to wake up from sleep

ravinsweden, 2020-12-12:
Hi,

I switched from version 4 to 6 and everything is working as expected, except  one thing. 

When I turn off the LG TV with the remote, the computer is suspended which is great. But after around 5-10 seconds the computer starts up and turning on the TV again. This only happens when running the Cec tray app, exiting the cec tray app and doing a manual suspend from windows start menu does not automatically wake the computer from suspend after 5-10 seconds. Is this a know bug with 6.0.2?

nibty, 2022-01-18:
I'm having the same issue. Were you able to find a solution? Downgrading works, but I have other issues with older versions.

ravinsweden, 2022-01-18:
Yes I solved it by disabling usb power settings in the bios. For example disabling power up by pressing any usb or pci device. It seems like the usb adapter is forcing the computer to power on like a keyboard. Please try to disable any support in the bios to power on through usb or pci.

closed 2025-02-27 by github-actions[bot]