Pulse-Eight/libcec #548: CEC only works if Pi is the only device connected to TV

gabaloliveira, 2020-12-17:
**Hardware**

- Raspberry Pi 1 Model A+
- Samsung UN49KS8000

**Software**
```
pi@RaspberryPi:~ $ cec-client -l
libCEC version: 6.0.2, git revision: libcec-6.0.2+1-76551ea, compiled on 2020-12-10 15:08:36 by pi@RaspberryPi on Linux 5.4.79+ (armv6l), features: P8_USB, DRM, P8_detect, randr, RPi
```

**Bug Description**

cec-client works just fine if the Raspberry Pi is the only HDMI device connected to the TV. However, it stops working instantly after plugging in any HDMI device, and any command after that returns a "timeout" debug message. 

All cec-client commands (power on, power off, change HDMI input, etc...) works with any HDMI cable I've tested so far, as long as the Pi is the only connected device. 

Starting cec-client having only connected the Raspberry Pi (brand new HDMI cable):
```
pi@RaspberryPi:~ $ cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect:
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [             402]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [             414]     logical address changed to Free use (e)
DEBUG:   [             419]     Open - vc_cec initialised
NOTICE:  [             422]     connection opened
DEBUG:   [             429]     << Broadcast (F) -> TV (0): POLL
DEBUG:   [             429]     initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [             429]     << e0
DEBUG:   [             428]     processor thread started
DEBUG:   [             462]     >> POLL sent
DEBUG:   [             470]     TV (0): device status changed into 'present'
DEBUG:   [             475]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [             478]     << e0:8c
TRAFFIC: [             670]     >> 0f:87:00:00:f0
DEBUG:   [             670]     TV (0): vendor = Samsung (0000f0)
DEBUG:   [             675]     >> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             681]     expected response received (87: device vendor id)
DEBUG:   [             681]     replacing the command handler for device 'TV' (0)
DEBUG:   [             682]     registering new CEC client - v6.0.2
DEBUG:   [             682]     SetClientVersion - using client version '6.0.2'
NOTICE:  [             682]     setting HDMI port to 1 on device TV (0)
DEBUG:   [             683]     SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [             683]     detecting logical address for type 'recording device'
DEBUG:   [             683]     trying logical address 'Recorder 1'
DEBUG:   [             683]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             683]     << 11
TRAFFIC: [             967]     << 11
DEBUG:   [            1238]     >> POLL not sent
DEBUG:   [            1238]     using logical address 'Recorder 1'
DEBUG:   [            1239]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            1239]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            1239]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            1239]     Recorder 1 (1): CEC version 1.4
DEBUG:   [            1239]     AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            1241]     logical address changed to Recorder 1 (1)
DEBUG:   [            1249]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            1249]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            1253]     GetPhysicalAddress - physical address = 4000
DEBUG:   [            1253]     using auto-detected physical address 4000
DEBUG:   [            1254]     Recorder 1 (1): physical address changed from ffff to 4000
DEBUG:   [            1254]     << Recorder 1 (1) -> broadcast (F): physical address 4000
TRAFFIC: [            1254]     << 1f:84:40:00:01
NOTICE:  [            1406]     CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 4.0.0.0, git revision: libcec-6.0.2+1-76551ea, compiled on 2020-12-10 15:08:36 by pi@RaspberryPi on Linux 5.4.79+ (armv6l), features: P8_USB, DRM, P8_detect, randr, RPi
DEBUG:   [            1406]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
[... 260 lines cut ...]
device #0: TV
address:       0.0.0.0
active source: yes
vendor:        Samsung
osd string:    TV
CEC version:   1.4
power status:  unknown
language:      por

device #1: Recorder 1
address:       4.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: TV (0)
standby 0 #does not work
NOTICE:  [          585472]     << putting 'TV' (0) in standby mode
TRAFFIC: [          585476]     << 10:36
TRAFFIC: [          585479]     << 10:36
```
I've already tried with 5+ different HDMI cables (bought 3 different cables just to be sure), power cycling the TV, unplugging the power cord for 3+ hours...

9000h, 2021-02-06:
the rpi 1 has a hardware issue
```
From http://www.raspberrypi.org/archives/1929 -
Some users have found that connecting an unpowered Raspberry Pi to an HDMI television interferes
with the correct operation of CEC for other connected devices. This was fixed on some later revision 1.0 PCBs
by removing the ESD protection diode D14; revision 2.0 fixes this issue by connecting the top side of the diode to +5V0_HDMI.
```

9000h, 2025-02-26:
It's still relevant to PI 1 Users
https://github.com/Pulse-Eight/libcec/issues/548#issuecomment-774444282
but there is nothing to fix in LibCEC, so a close should be fine.

opdenkamp, 2025-02-27:
If the fix is removing that diode, then libCEC can't help indeed.
Thanks for reporting back!

closed 2025-02-27 by opdenkamp