Pulse-Eight/libcec #554: Failing to switch off LG OLED TV 55E9

bk1285, 2021-01-21:
Hi all,

unfortunetly I'm failing to switch off my LG OLED TV 55E9. Switching it on is not a problem, but the command below is somehow ignored by the TV (is still switched on afterwards):

```
$ echo "standby 0" | ./cec-client.exe -s
opening a connection to the CEC adapter...
DEBUG:   [               2]     Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              39]     connection opened, clearing any previous input and waiting for active transmissions to end before starting
DEBUG:   [              39]     communication thread started
DEBUG:   [              50]     turning controlled mode on
NOTICE:  [              73]     connection opened
DEBUG:   [              73]     processor thread started
DEBUG:   [              73]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [              73]     << f0
DEBUG:   [              73]     setting the line timeout to 3
DEBUG:   [             114]     >> POLL sent
DEBUG:   [             114]     TV (0): device status changed into 'present'
DEBUG:   [             114]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [             114]     << f0:8c
DEBUG:   [            1171]     expected response not received (87: device vendor id)
TRAFFIC: [            1171]     << f0:8c
DEBUG:   [            2227]     expected response not received (87: device vendor id)
NOTICE:  [            2227]     registering new CEC client - v4.0.2
DEBUG:   [            2227]     detecting logical address for type 'recording device'
DEBUG:   [            2227]     trying logical address 'Recorder 1'
DEBUG:   [            2227]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            2227]     << 11
DEBUG:   [            2257]     CEC transmission - received response - TRANSMIT_FAILED_ACK
TRAFFIC: [            2257]     << 11
DEBUG:   [            2292]     CEC transmission - received response - TRANSMIT_FAILED_ACK
DEBUG:   [            2292]     >> POLL not sent
DEBUG:   [            2292]     using logical address 'Recorder 1'
DEBUG:   [            2292]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            2292]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            2292]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            2292]     Recorder 1 (1): CEC version 1.4
DEBUG:   [            2292]     CEC::CCECClient::AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            2292]     setting ackmask to  2
DEBUG:   [            2297]     Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            2297]     Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            2297]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [            2298]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - ADL returned physical address 0000
DEBUG:   [            2298]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via nvidia driver
DEBUG:   [            2298]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - nvidia driver returned physical address 0000
DEBUG:   [            2298]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address from the OS
DEBUG:   [            2299]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - OS returned physical address 0000
NOTICE:  [            2299]     setting HDMI port to 1 on device TV (0)
DEBUG:   [            2299]     Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [            2299]     << Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [            2299]     << 1f:84:10:00:01
DEBUG:   [            2433]     using persisted autonomous mode setting: 'enabled'
WARNING: [            2439]     COMMAND_REJECTED: GET_HDMI_VERSION
TRAFFIC: [            2530]     >> 01:8f
DEBUG:   [            2530]     << Recorder 1 (1) -> TV (0): on
TRAFFIC: [            2530]     << 10:90:00
DEBUG:   [            2530]     >> TV (0) -> Recorder 1 (1): give device power status (8F)
TRAFFIC: [            2731]     >> 01:1a:01
DEBUG:   [            2731]     << Recorder 1 (1) -> TV (0): deck status 'stop'
TRAFFIC: [            2731]     << 10:1b:1a
DEBUG:   [            2731]     >> TV (0) -> Recorder 1 (1): give deck status (1A)
DEBUG:   [            3434]     command 'GET_HDMI_VERSION' was not acked by the controller
DEBUG:   [            3438]     using persisted logical address setting: 'Recorder 1'
DEBUG:   [            3443]     using persisted device type setting: 'recording device'
DEBUG:   [            3449]     using persisted logical address mask setting: '206'
DEBUG:   [            3467]     using persisted device name setting: 'CEC Adapter'
DEBUG:   [            3472]     using persisted physical address setting: '1000'
NOTICE:  [            3472]     CEC client registered: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 12, firmware build date: Tue Apr 28 14:20:49 2020 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, compiled on Windows-6.3.9600, features: P8_USB, P8_detect
DEBUG:   [            3472]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            3472]     << 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            3745]     << requesting power status of 'TV' (0)
TRAFFIC: [            3745]     << 10:8f
DEBUG:   [            4804]     expected response not received (90: report power status)
TRAFFIC: [            4804]     << 10:8f
DEBUG:   [            5862]     expected response not received (90: report power status)
DEBUG:   [            5862]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [            5862]     << 10:8c
TRAFFIC: [            6826]     >> 0f:87:00:e0:91
DEBUG:   [            6826]     TV (0): vendor = LG (00e091)
DEBUG:   [            6826]     << Recorder 1 (1) -> Broadcast (F): vendor id Pulse Eight (1582)
TRAFFIC: [            6826]     << 1f:87:00:15:82
DEBUG:   [            6826]     >> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [            6921]     expected response not received (87: device vendor id)
TRAFFIC: [            6956]     << 10:8c
DEBUG:   [            7019]     expected response received (87: device vendor id)
DEBUG:   [            7019]     replacing the command handler for device 'TV' (0)
DEBUG:   [            7019]     TV (0): CEC version 1.3a
DEBUG:   [            7019]     TV (0): menu language set to 'eng'
DEBUG:   [            7019]     Recorder 1 (1): vendor = LG (00e091)
DEBUG:   [            7019]     replacing the command handler for device 'Recorder 1' (1)
DEBUG:   [            7019]     Recorder 1 (1): CEC version 1.3a
NOTICE:  [            7019]     << putting 'TV' (0) in standby mode
TRAFFIC: [            7019]     << 10:36
DEBUG:   [            7076]     unregistering all CEC clients
NOTICE:  [            7076]     unregistering client: libCEC version = 4.0.2, client version = 4.0.2, firmware version = 12, firmware build date: Tue Apr 28 14:20:49 2020 +0000, logical address(es) = Recorder 1 (1) , base device: TV (0), HDMI port number: 1, physical address: 1.0.0.0, compiled on Windows-6.3.9600, features: P8_USB, P8_detect
DEBUG:   [            7076]     Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [            7076]     Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [            7076]     Recorder 1 (1): CEC version unknown
DEBUG:   [            7076]     Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [            7076]     Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [            7076]     setting ackmask to  0
DEBUG:   [            7081]     turning controlled mode off
DEBUG:   [            7087]     unregistering all CEC clients
DEBUG:   [            8082]     communication thread ended

```

I'd appreciate any feedback pointing me towards a solution... :)

Best,
Bernd

filimonic, 2021-04-10:
As noted [on support matrix](http://libcec.pulse-eight.com/vendor/support), LG TVs do not support PowerOff feature. ( I own two TVs - Y2011 and Y2014 models ). There is no fw updates from LG that fix it. 

Newer models, as far I know from others, support this feature completely.

filimonic, 2023-01-12:
Hardware workaround for LG TVs with WebOS not supporting CEC STANDBY https://github.com/ALFISYS/cec_lg_off

filimonic, 2025-02-26:
@opdenkamp Maybe close this as not possible to implement as LG TVs prior ~2015 year models do not support standby signal from CEC ?

opdenkamp, 2025-02-27:
Yeah if they don't support CEC standby, then libCEC can't help indeed. Thanks for posting your hardware work around.

closed 2025-02-27 by opdenkamp