Pulse-Eight/libcec #564: How do I determine if my hardware is supported?

penguinairlines, 2021-04-26:
I am looking to enable CEC controls on a NUC style device that I have, but I have installed cec-utils and I see that `cec-client -l` returns `Found devices: NONE`. I figure this is probably because I am not using supported hardware, but I looked through the documentation and I do not see anything to indicate hardware checking commands in the documentation or FAQ, but maybe I missed it as I only spent about an hour looking. 

Some metadata that may be incidentally helpful:
libCEC version: 4.0.4
Linux-4.4.0-176 (Pop!_OS aka Debian)

malard, 2021-04-26:
You need our cec adapter, visit www.pulse-eight.com

penguinairlines, 2021-04-26:
Oh! I understand now, thank you. I was seeing that others could use this package without dedicated hardware, such on a Raspberry Pi, is that correct? Or is there another similarly named package?

malard, 2021-04-27:
We support the pi natively

Stevenragy, 2023-01-25:
@malard I'm using ASUS tinker board 2s and I'm trying to run `cec-client -l` and it returns Found devices: NONE 
Also by using cec-ctl it works correctly but using cec-client it doesn't 
Do you have any idea how should i setup my device?

I'm using tinker OS debian 10

malard, 2023-01-25:
you need our hardware

EngineerGuy314, 2023-01-29:
just to clarify, we need dedicated CEC hardware, *or* a Raspberry Pi which is "natively" supported?

But are any other SBC's natively supported? Specifically the OrangePi?

malard, 2023-01-29:
Either our software needs to support the native hardware that has the cec wire connected, or you need our adapter

malard, 2023-02-14:
I have no idea sorry, I’ve never heard of or seen this hardware, you may need to ask them how they do it?

I am confused. Libreelec uses libcec and with the newer versions CEC is working on the Orange Pi (Allwinner H6).

OrangePi/Allwinner are NOT listed on the list of native supported hardware. So is the list of supported hardware on libcec need to be updated? Or is Libreelec using some non-libcec method to do CEC on Orange Pi?

closed 2025-02-27 by github-actions[bot]