Pulse-Eight/libcec #565: libcec settings to disable executing incoming turn off signals

mghadam, 2021-05-05:
I have a Raspberry PI 4 that is connected to a Samsung TV.

Whenever I turn off the TV, it also turns the Pi off. This is while I want to keep the Pi on all the time.

If I disable this feature called "Auto Standby" in the TV, I will not be able to turn off the TV from the Pi.

So the only way to be able to turn off the TV from the Pi, is to keep the "Auto Standby" setting active in the TV. 

Is it possible to configure libcec on to avoid executing turn off signals that it receives from the TV?

popcornmix, 2021-05-05:
Have you changed CEC setting "When the TV is switched off" to "Ignore"?
In System/Input/Peripherals/CEC Adapter.

mghadam, 2021-05-05:
No, it is running Raspberry Pi OS (Raspbian).

Kodi has the feature you mentioned. However, I can not find how to set this setting in Raspbian.

popcornmix, 2021-05-06:
Raspbian won't do anything with CEC if you don't install a package that uses it (kodi is the main one).
Are you sure it's CEC? Are you powering the Pi from a USB port on TV?

mghadam, 2021-05-06:
Yes it is libcec on raspberry pi :

```
root@raspberrypi:~# dpkg -l | grep -i cec
ii  cec-utils                            4.0.4+dfsg1-2+rpi1+rpt1                 armhf        USB CEC Adaptor communication Library (utility programs)
ii  libcec4:armhf                        4.0.4+dfsg1-2+rpi1+rpt1                 armhf        USB CEC Adaptor communication Library (shared library)
root@raspberrypi:~# 

```

The Pi has its own adapter and it is not powered from the USB ports on the TV.

I use cec-utils to [control the TV](https://pimylifeup.com/raspberrypi-hdmi-cec/).

libcec supports CEC on the Pi via a [VCHIQ service](https://github.com/Pulse-Eight/libcec/tree/master/src/libcec/adapter/RPi)  ([Source](https://www.raspberrypi.org/forums/viewtopic.php?t=259557)). I checked the codes there and it seems that the signals raspberry receives from the TV are sent to libcec for action:

 ```
 case VC_CEC_RX:
    // CEC data received
    {
      // translate into a VC_CEC_MESSAGE_T
      VC_CEC_MESSAGE_T message;
      vc_cec_param2message(header, p0, p1, p2, p3, &message);

      // translate to a cec_command
      cec_command command;
      cec_command::Format(command,
          (cec_logical_address)message.initiator,
          (cec_logical_address)message.follower,
          (cec_opcode)CEC_CB_OPCODE(p0));

      // copy parameters
      for (uint8_t iPtr = 1; iPtr < message.length; iPtr++)
        command.PushBack(message.payload[iPtr]);

      // send to libCEC
      m_callback->OnCommandReceived(command);
    }

```

So do I need to configure libcec to ignore incoming Poweroff signals (`:36`)?

popcornmix, 2021-05-06:
What code do you believe actually powers off the Pi?

mghadam, 2021-05-06:
I think it is the TV that sends a ":36" signal via HDMI-CEC to the PI whenever the tv is turned off, and this turns off the Pi. This only happens when the "Auto Standby" feature is enabled in the TV. 

I can disable "Auto Standby" in the TV to stop this behavior however this will also prevents the pi to turn off the TV as well i.e. the TC will ignore ":36" signals sent from the Pi. (See note #2 in Samsung row [here](http://libcec.pulse-eight.com/Vendor/Support)).

I want to be able to turn off the TV from the Pi, without the Pi getting turned off when the TV sends a turn off signal.

popcornmix, 2021-05-06:
> and this turns off the Pi

As far as I know libcec only passes the message on that TV hasentered standby.
An app (like kodi) can choose what to do with that (it has a menu option with choices like "ignore/pause video/stop video/suspend/power off"). But that is something kodi does rather than libcec.

I'm not aware of cec-client doing that. I wondered if you had some other cec app that did it.

mghadam, 2021-05-23:
I could not reproduce this issue. 

I enabled "autostandby" in the samsung TV and can turn it off / on using HDMI-CEC, and when I turn the TV off, the raspberry pi remains on which is what I was looking for.

Anyways, I found CEC_DEFAULT_SETTING_POWER_OFF_ON_STANDBY and bPowerOffOnStandby settings [here](url=https://github.com/Pulse-Eight/libcec/blob/master/include/cectypes.h) for future references.

closed 2021-05-23 by mghadam