Pulse-Eight/libcec #566: CEC Bit timing configurationÉ

drsamko, 2021-05-06:
Hi there,

we're using LibCEC on a Pi 4 to verify our AVR for certification. One requirement is to test the support for the full range of bit timing (e.g. Logical '1' low level between .4 and .8ms etc.),

Currently the controller is using the standard values (i.e. .6ms for the example above). Is there a way to configure the CEC driver and update the bit timing? I'm particularly looking into sending an <Abort> message with Logical '1' Low timing length of 400us, as well as another test with Logical '1' High timing length of 1450us.

Any help is immensely appreciated!

malard, 2021-05-06:
I don’t think you can do this because the timing is controlled at a SoC level on the Pi, however in theory this could be done on our external adapters but would need custom firmware to achieve this.

Currently the controller is using the standard values (i.e. .6ms for the example above). Is there a way to configure the CEC driver and update the bit timing? I'm particularly looking into sending an message with Logical '1' Low timing length of 400us, as well as another test with Logical '1' High timing length of 1450us.

drsamko, 2021-05-06:
Thanks for the quick response, @malard, are there any public resources regarding updating the firmware on the external device (can we do it in-house)? We could potentially look into that if it can't be done on the Pi.

malard, 2021-05-06:
Please email ***@***.*** as it would be a commercial effort

Thanks

drsamko, 2021-05-06:
Thanks, though your email is masked in your message. can you separate it please? We might possibly approach
If it's your email, I can grab it from your profile

malard, 2021-05-06:
Sales at pulse-eight dot com

popcornmix, 2021-05-07:
If you use the kms driver on Pi and build libcec for linux (`-DHAVE_LINUX_API=1 -DHAVE_RPI_API=0`)
then the driver is available to change. I suspect you want to start looking here:
https://github.com/raspberrypi/linux/blob/53a5ac4935c500d32bfc465551cc5107e091c09c/drivers/gpu/drm/vc4/vc4_hdmi.c#L2270

drsamko, 2021-05-07:
Thanks @popcornmix , looks promising!

closed 2025-02-27 by github-actions[bot]