Pulse-Eight/libcec #567: libCEC continuously disconnects from LG OLED TV making impossible to use

ignisf, 2021-05-17:
#### Hardware:

* Khadas VIM1
* LG OLED 65BX3LB (2020 model afaik), latest available WebOS

#### Software:

Freshly installed CoreELEC 19.1-Matrix, though the issue was present on 9.2.7 as well

libCEC version: 6.0.2,  compiled on 2021-05-12 11:26:34 by portisch@ubuntu on Linux 5.4.0-73-generic (x86_64), features: P8_USB, DRM, P8_detect, AOCEC

#### Full debug log:
https://gist.github.com/ignisf/f90343b03b55fe79d66bf6ce286ea6f3

#### Observed issue:

The CEC functionality of the TV gets disabled every around 30 seconds (the remote stops being able to control Kodi) and is gone for around 10 seconds, then reappears.

The logs are full with power status changes `from 'on' to 'in transition from standby to on'` and `FIXME: LG seems to have bugged out. resetting to 'in transition standby to on'. the return button will not work`.

Rarely, after a random sequence of reboots, I am able to get a stable connection. In this case the logs indicate that libCEC has received the Recorder 2 (2) logical address. Also the 'Title' entry is missing from the CEC menu on the TV.

Relevant excerpt (the device disappears right after `2021-05-17 00:18:53.911 T:4739    DEBUG <general>: CecLogMessage - << POLL: TV (0) -> Recorder 1 (1)`:
```
2021-05-17 00:18:42.394 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): OSD name 'CoreELEC'
2021-05-17 00:18:42.394 T:4739    DEBUG <general>: CecLogMessage - << 10:47:43:6f:72:65:45:4c:45:43
2021-05-17 00:18:42.394 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): give osd name (46)
2021-05-17 00:18:42.744 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:0b
2021-05-17 00:18:42.744 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'on' to 'in transition from standby to on'
2021-05-17 00:18:42.744 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): in transition from standby to on
2021-05-17 00:18:42.744 T:4739    DEBUG <general>: CecLogMessage - << 10:90:02
2021-05-17 00:18:42.744 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:18:42.839 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'in transition from standby to on' to 'on'
2021-05-17 00:18:42.928 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:01
2021-05-17 00:18:42.928 T:4739    DEBUG <general>: CecLogMessage - << 10:89:02:05
2021-05-17 00:18:42.928 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:18:43.047 T:4739    DEBUG <general>: CecLogMessage - SL initialised
2021-05-17 00:18:53.730 T:4739    DEBUG <general>: CecLogMessage - >> 0f:87:00:e0:91
2021-05-17 00:18:53.731 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> Broadcast (F): vendor id LG (e091)
2021-05-17 00:18:53.731 T:4739    DEBUG <general>: CecLogMessage - << 1f:87:00:e0:91
2021-05-17 00:18:53.731 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Broadcast (F): device vendor id (87)
2021-05-17 00:18:53.911 T:4739    DEBUG <general>: CecLogMessage - >> 01
2021-05-17 00:18:53.911 T:4739    DEBUG <general>: CecLogMessage - << POLL: TV (0) -> Recorder 1 (1)
2021-05-17 00:18:54.004 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:0b
2021-05-17 00:18:54.004 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'on' to 'in transition from standby to on'
2021-05-17 00:18:54.004 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): in transition from standby to on
2021-05-17 00:18:54.004 T:4739    DEBUG <general>: CecLogMessage - << 10:90:02
2021-05-17 00:18:54.004 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:18:54.097 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'in transition from standby to on' to 'on'
2021-05-17 00:19:07.730 T:4739    DEBUG <general>: CecLogMessage - >> 0f:87:00:e0:91
2021-05-17 00:19:07.731 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> Broadcast (F): vendor id LG (e091)
2021-05-17 00:19:07.731 T:4739    DEBUG <general>: CecLogMessage - << 1f:87:00:e0:91
2021-05-17 00:19:07.731 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Broadcast (F): device vendor id (87)
2021-05-17 00:19:07.911 T:4739    DEBUG <general>: CecLogMessage - >> 01
2021-05-17 00:19:07.911 T:4739    DEBUG <general>: CecLogMessage - << POLL: TV (0) -> Recorder 1 (1)
2021-05-17 00:19:08.006 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:0b
2021-05-17 00:19:08.006 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'on' to 'in transition from standby to on'
2021-05-17 00:19:08.006 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): in transition from standby to on
2021-05-17 00:19:08.006 T:4739    DEBUG <general>: CecLogMessage - << 10:90:02
2021-05-17 00:19:08.006 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:19:08.100 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'in transition from standby to on' to 'on'
2021-05-17 00:19:08.188 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:01
2021-05-17 00:19:08.188 T:4739    DEBUG <general>: CecLogMessage - << 10:89:02:05
2021-05-17 00:19:08.188 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:19:08.308 T:4739    DEBUG <general>: CecLogMessage - SL initialised
2021-05-17 00:19:08.372 T:4739    DEBUG <general>: CecLogMessage - >> 01:83
2021-05-17 00:19:08.372 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> broadcast (F): physical address 4000
2021-05-17 00:19:08.372 T:4739    DEBUG <general>: CecLogMessage - << 1f:84:40:00:01
2021-05-17 00:19:08.372 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): give physical address (83)
2021-05-17 00:19:08.604 T:4739    DEBUG <general>: CecLogMessage - >> 01:46
[... 6 lines cut ...]
2021-05-17 00:19:08.929 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): give device vendor id (8C)
2021-05-17 00:19:09.134 T:4739    DEBUG <general>: CecLogMessage - >> 01:8f
2021-05-17 00:19:09.134 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): on
2021-05-17 00:19:09.134 T:4739    DEBUG <general>: CecLogMessage - << 10:90:00
2021-05-17 00:19:09.134 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): give device power status (8F)
2021-05-17 00:19:09.291 T:4739    DEBUG <general>: CecLogMessage - >> 01:8f
2021-05-17 00:19:09.291 T:4739  WARNING <general>: CecLogMessage - FIXME: LG seems to have bugged out. resetting to 'in transition standby to on'. the return button will not work
2021-05-17 00:19:09.291 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'on' to 'in transition from standby to on'
2021-05-17 00:19:09.291 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): in transition from standby to on
2021-05-17 00:19:09.291 T:4739    DEBUG <general>: CecLogMessage - << 10:90:02
2021-05-17 00:19:09.291 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): give device power status (8F)
2021-05-17 00:19:09.382 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'in transition from standby to on' to 'on'
2021-05-17 00:19:09.471 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:0b
2021-05-17 00:19:09.471 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'on' to 'in transition from standby to on'
2021-05-17 00:19:09.471 T:4739    DEBUG <general>: CecLogMessage - << Recorder 1 (1) -> TV (0): in transition from standby to on
2021-05-17 00:19:09.471 T:4739    DEBUG <general>: CecLogMessage - << 10:90:02
2021-05-17 00:19:09.471 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:19:09.566 T:4739    DEBUG <general>: CecLogMessage - Recorder 1 (1): power status changed from 'in transition from standby to on' to 'on'
2021-05-17 00:19:09.657 T:4739    DEBUG <general>: CecLogMessage - >> 01:89:01
2021-05-17 00:19:09.657 T:4739    DEBUG <general>: CecLogMessage - << 10:89:02:05
2021-05-17 00:19:09.657 T:4739    DEBUG <general>: CecLogMessage - >> TV (0) -> Recorder 1 (1): vendor command (89)
2021-05-17 00:19:09.774 T:4739    DEBUG <general>: CecLogMessage - SL initialised
2021-05-17 00:19:09.839 T:4739    DEBUG <general>: CecLogMessage - >> 01:8f
2021-05-17 00:19:09.839 T:4739  WARNING <general>: CecLogMessage - FIXME: LG seems to have bugged out. resetting to 'in transition standby to on'. the return button will not work
```

ignisf, 2021-05-17:
OK, I am able to reproduce the issue reliably with `cec-client -d 0` (https://gist.github.com/ignisf/1c5831f87a1ef812e80c86676c47b76b).

The issue is not present when invoking `cec-client -t p -d 0` (https://gist.github.com/ignisf/af0115eb3e198438ffc6edf1fe01fa7c).

Workaround for Kodi:

1. Stop Kodi (`systemctl stop kodi` on CoreELEC).
2. Set `device_type` in `userdata/peripheral_data/cec_CEC_Adapter.xml` to "4".
3. Restart Kodi.

opdenkamp, 2025-02-27:
This would be something for the third party application to add as an option in that case. It's not a generic LG issue and changing the default to playback device for all LG users could cause trouble with other models. It does detect that there is a problem in the log (`FIXME: LG seems to have bugged out. resetting to 'in transition standby to on'. the return button will not work`) but LG also gets into this state when an invalid physical address is used. So unfortunately this is not something that libCEC can autotically address.

ignisf, 2025-02-27:
Also, if indeed there is no way to discern a specific TV model/version to apply a quirk, then the only thing that can be done is to document as known issue and specify workaround to try 🤷 .

closed 2025-05-29 by github-actions[bot]