Pulse-Eight/libcec #573: How to find codes?

penguinjeff, 2021-06-19:
I found a few codes searching with google but is seems difficult to find codes to send.
Codes I found so far

#mute reciever
alias m='echo "tx 15:44:43" | cec-client -s'
#turn off reciever
alias off='echo "tx 15:36" | cec-client -s'
#turn on receiver
alias on='echo "tx 15:44:40" | cec-client -s'
#request power status of receiver
alias pwstat='if [ "$(echo "tx 15:8f" | cec-client -s|grep ">>"|sed "s/.*>> //"|grep 51:90)" = "51:90:00" ];then echo on;else echo off; fi '
#change input of receiver
alias t='echo "tx 15:44:69" | cec-client -s'
#change the input of receiver to the tv input (toslink)
alias tv='echo "tx 4F:82:30:00" | cec-client -s'
#raise receiver volume
alias v+='echo "tx 15:44:41" | cec-client -s'
#lower receiver volume
alias v-='echo "tx 15:44:42" | cec-client -s'
#denon BD
alias HDMI1='echo "tx 4F:82:11:00" | cec-client -s'
#denon DVD
alias HDMI2='echo "tx 4F:82:12:00" | cec-client -s'
#denon SAT
alias HDMI3='echo "tx 4F:82:13:00" | cec-client -s'
#denon GAME
alias HDMI4='echo "tx 4F:82:14:00" | cec-client -s'
alias down='echo "tx 15:44:06"| cec-client -s'
alias up='echo "tx 15:44:01"| cec-client -s'
alias menu='echo "tx 15:44:0A"| cec-client -s'
alias menu_exit='echo "tx 15:44:0D"| cec-client -s'

malard, 2021-06-19:
Try cec-o-matic.com

cheers

penguinjeff, 2021-06-19:
> Try cec-o-matic.com cheers From: penguinjeff ***@***.***> Sent: 19 June 2021 20:07 To: Pulse-Eight/libcec ***@***.***> Cc: Subscribed ***@***.***> Subject: [Pulse-Eight/libcec] How to find codes? (#573)

I could not figure out how to use that site. It doesn't seem to work right.

malard, 2021-06-20:
The specification explains it also, the cec-o-matic website works correctly

penguinjeff, 2021-06-20:
Is there a tutorial on how to use it? What do I type in where and how do I use any output. I don't understand how the output translates to hex codes I can send to cec-client. How do I get the information to type into cec-o-matic? Do I type in the hex of the manufacture? or the hex of the device? Or join them together or what? After I type in the code how do I find supported hex codes that I can send to the device what do I click next?

penguinjeff, 2021-06-20:
The "try this" puts a code "F0:64:00:48:65:6C:6C:6F:20:77:6F:72:6C:64" but doesn't explain what that is?

malard, 2021-06-20:
Press the down arrow to see the result

penguinjeff, 2021-06-20:
If I click on the arrow after it shows source "unregistered" destination "TV" What is source and what is destination? It looks like it probably has the text "hello world" in the hex and that this is probably the code to set the On Screen Display(OSD) to "hello world" maybe on a tv? Where would the OSD even be on a TV? So would it only show me commands I already know? That is not what I was asking for I want to know where I can get the commands I do not know?

penguinjeff, 2021-06-20:
OK I think I figured it out a little.  It seems a bit incomplete. Now I know a bit more about what the command it is that I am sending.
And that I can modify the command to be more accurate. Thank you for your help.
This might help me build a few of my commands. I guess denon receivers do not follow some of the standards. I'm not sure how to change the channel on my radio with this.

DunkanA, 2021-06-20:
I've also struggled with cec-o-matic, but have figured out that some
commands get sent and it doesn't matter where they come from, so you just
say they come from Unregistered.  If you enter all the commands you know,
and press the down arrow, they will explain what CEC command is being
issued.  If you modify the command, i.e. change the source or destination,
then press the up arrow to get the new code.  You can try all the CEC
commands and sometimes cec-client will reply saying it's unsupported.  Or
it, accepted and doesn't work.

If you study the format of your known commands you will see that the 1st
digits is the source (0-F = 0-15) and the 2nd digit is the destination.
The 3rd/4th digits are the command to issue, the 5/6th are the subcommand,
followed by any parameters/text required

You asked for all the commands you don't know, well cec-o-matic has all the
codes that CEC standard supports, you might have to talk to the vendors of
your kit to see if they will give you a list of what commands they support.
--
Dunkan Armstrong
Edinburgh, Scotland

penguinjeff, 2021-06-20:
Yeah I just figured out the commands still work from unregistered. I will write up some instructions for cec-o-matic as I want even less struggle for the next guy. I still doubt it will show me how to use my radio on my denon receiver but maybe looking up what othes have done now that I know some commands will give me a clue where to start.

penguinjeff, 2021-06-20:
Ooooh with typing in the F5:44:02 command I see a good list of command that I wanted to know now to try them out and see if they work on my receiver.

closed 2025-02-27 by github-actions[bot]