Pulse-Eight/libcec #576: HDMI-CEC suddenly stopped working on Samsung TV

mghadam, 2021-08-01:
I have a Samsung TV with RPI 4 connected to it. The HDMI-CEC was working decently for a couple of months until last night when it suddenly stopped working.

Whenever I reboot the raspberry pi, I can see that RPI sees the TV only in the first scan, then it does not show the TV again, this is the output of scan after rebooting the raspberry pi :
```
root@raspberrypi:~# echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   unknown
power status:  unknown
language:      ???

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```
Now I run the scan again:
```
root@raspberrypi:~# echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
root@raspberrypi:~# 
```
I can see RPi desktop on my TV without any issues when I manually switch the TV to HDMI. 

Here is the fixes I tried: 
- unplugged everything and plugged them again after 10mins and it didn't help.
- removed rpi libcec and cec-client and installed the latest version of libcec v6 and it's the same
- added `hdmi_force_hotplug=1` to /boot/config.txt and still no dice

mghadam, 2021-08-01:
I downgraded the rpi kernel from `v1:1.20210727-1` to `v1:1.20210527-1` and this was resolved. Looks like it was caused by [this commit](https://github.com/raspberrypi/firmware/commit/775e6a7eb1c5630d9e118c250a15614b1be30e0e) in RaspberryPi Kernel...

vendetta1987, 2021-08-04:
I seem to have the exact same problem in relation to Kodi. Its CEC-feature stopped working after I updated my system by apt update. I can also reproduce your scan issue apart from Kodi. However downgrading to [ab72b99](https://github.com/Hexxeh/rpi-firmware/tree/ab72b99243adad8950325c948b4b57ca84427a08) from the stable branch via rpi-update doesn't fix it for me. How exactly did you perfom the mentioned downgrade? The version numbers are related to the `raspberrypi-kernel` package as far as I can tell. Mine doesn't seem to find any other version than `v1:1.20210727-1`

mghadam, 2021-08-05:
> I seem to have the exact same problem in relation to Kodi. Its CEC-feature stopped working after I updated my system by apt update. I can also reproduce your scan issue apart from Kodi. However [...]

I had the packages in apt cache and downgraded them using this command:

```
dpkg -i /var/cache/apt/archives/libraspberrypi-doc_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/libraspberrypi-dev_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/raspberrypi-kernel_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/libraspberrypi-bin_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/libraspberrypi0_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/raspberrypi-bootloader_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/linux-libc-dev_1%3a1.20210527-1_armhf.deb \
/var/cache/apt/archives/raspberrypi-kernel-headers_1%3a1.20210527-1_armhf.deb
```

Please also mention your issue on the [RPi kernel commit issue](https://github.com/raspberrypi/linux/issues/4460) that caused this problem so the devs know there are more users that were affected by the patch.

vendetta1987, 2021-08-10:
The latest apt (kernel) update fixed this for me.

isavcic, 2021-08-29:
Running rpi-update without arguments bumped the firmware to 895ca75fe52090a2893725c2e752cce86aeb227f and kernel to 5.10.60. Running consecutive cec-client scans detects the Samsung TV, but the vendor string is missing in consecutive scans, I get the weird error `ERROR:   [           99540]     failed to request the physical address`, and a spurious Audio device appears to which Kodi keeps connecting and effectively ignores the Samsung remote control.

isavcic, 2021-08-29:
@mghadam which kernel version is that (uname -a)?

isavcic, 2021-08-29:
@mghadam it's symptomatic that you don't see the CEC version; do you have the HDMI connected to port #0 (further away from power connector) by any chance? If so, connect to #1.

mghadam, 2021-08-30:
It was a kernel bug and resolved after they released a new version of the kernel.

I have no problem with cec now :

```
root@raspberrypi:~# echo scan | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   unknown
power status:  standby
language:      per

device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
root@raspberrypi:~# 
```

closed 2021-08-30 by mghadam