Pulse-Eight/libcec #577: Segmentation fault - raspberry 4

sfoglia, 2021-08-05:
Basicaly I'm trying to debug why kodi cec support seems to not work anymore with my TV and ended here.

HW: raspberry pi 4 4GB 
SW: raspberry pi os

Linux PItv 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux

I've followed: [libcec/README.raspberrypi.md at master · Pulse-Eight/libcec · GitHub](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md)
```

    sudo apt-get update
    sudo apt-get install cmake libudev-dev libxrandr-dev python3-dev swig git
    cd
    git clone https://github.com/Pulse-Eight/platform.git
    mkdir platform/build
    cd platform/build
    cmake ..
    make
    sudo make install
    cd
    git clone https://github.com/Pulse-Eight/libcec.git
    mkdir libcec/build
    cd libcec/build
    cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
    make -j4
    sudo make install
    sudo ldconfig
    sudo reboot
```
Now if I start:
`  sudo cec-client -t p
`
result in:
```
== using device type 'playback device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect: 
 path:     Raspberry Pi
 com port: RPI

opening a connection to the CEC adapter...
DEBUG:   [              73]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              73]	logical address changed to Free use (e)
DEBUG:   [              73]	Open - vc_cec initialised
NOTICE:  [              73]	connection opened
DEBUG:   [              74]	processor thread started
DEBUG:   [              74]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [              74]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              74]	<< e0
DEBUG:   [              74]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              74]	<< e0
DEBUG:   [              75]	>> POLL not sent
DEBUG:   [              75]	TV (0): device status changed into 'not present'
DEBUG:   [              75]	registering new CEC client - v6.0.2
DEBUG:   [              75]	SetClientVersion - using client version '6.0.2'
NOTICE:  [              75]	setting HDMI port to 1 on device TV (0)
DEBUG:   [              75]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [              75]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              75]	<< e0
DEBUG:   [              75]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              75]	<< e0
DEBUG:   [              75]	>> POLL not sent
DEBUG:   [              75]	SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [              75]	detecting logical address for type 'playback device'
DEBUG:   [              77]	trying logical address 'Playback 1'
DEBUG:   [              77]	<< Playback 1 (4) -> Playback 1 (4): POLL
TRAFFIC: [              77]	<< 44
TRAFFIC: [            1609]	<< 44
DEBUG:   [            3142]	>> POLL not sent
DEBUG:   [            3143]	using logical address 'Playback 1'
DEBUG:   [            3143]	Playback 1 (4): device status changed into 'handled by libCEC'
DEBUG:   [            3143]	Playback 1 (4): power status changed from 'unknown' to 'on'
DEBUG:   [            3143]	Playback 1 (4): vendor = Pulse Eight (001582)
DEBUG:   [            3143]	Playback 1 (4): CEC version 1.4
DEBUG:   [            3143]	logical address changed to Broadcast (f)
DEBUG:   [            3143]	AllocateLogicalAddresses - device '0', type 'playback device', LA '4'
DEBUG:   [            4676]	logical Broadcast (f) address lost
NOTICE:  [            4676]	logical address f was taken by another device, allocating a new address
DEBUG:   [            4676]	TV (0): device status changed into 'unknown'
DEBUG:   [            4676]	<< Playback 1 (4) -> TV (0): POLL
TRAFFIC: [            4676]	<< 40
DEBUG:   [            5677]	command 'feature abort' timeout
TRAFFIC: [            6177]	<< 40
DEBUG:   [            6209]	sending command 'feature abort' failed (4)
DEBUG:   [            6209]	>> POLL not sent
DEBUG:   [            6209]	TV (0): device status changed into 'not present'
DEBUG:   [            6209]	logical address changed to Playback 1 (4)
WARNING: [            6209]	unhandled response received: opcode=0 initiator=4 destination=0 response=3
DEBUG:   [            6209]	logical address changed to Broadcast (f)
DEBUG:   [            6209]	logical Broadcast (f) address lost
NOTICE:  [            6209]	logical address f was taken by another device, allocating a new address
DEBUG:   [            6209]	detecting logical address for type 'playback device'
DEBUG:   [            6209]	trying logical address 'Playback 2'
DEBUG:   [            6209]	<< Playback 2 (8) -> Playback 2 (8): POLL
DEBUG:   [            6209]	TV (0): device status changed into 'unknown'
TRAFFIC: [            6209]	<< 88
WARNING: [            6210]	cannot find the primary device (logical address f)
ERROR:   [            6210]	failed to register the CEC client: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, base device: TV (0), HDMI port number: 1, physical address: f.f.f.f, git revision: libcec-6.0.2+1-76551ea, compiled on 2021-08-04 15:09:11 by pi@PItv on Linux 5.10.52-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
DEBUG:   [            6210]	<< Broadcast (F) -> TV (0): OSD name 'Broadcast'
TRAFFIC: [            7741]	<< 88
DEBUG:   [            9275]	>> POLL not sent
DEBUG:   [            9275]	using logical address 'Playback 2'
DEBUG:   [            9275]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [            9275]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [            9275]	<< e0
DEBUG:   [            9275]	Playback 2 (8): device status changed into 'handled by libCEC'
DEBUG:   [            9275]	Playback 2 (8): power status changed from 'unknown' to 'on'
DEBUG:   [            9275]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [            9275]	Playback 2 (8): vendor = Pulse Eight (001582)
DEBUG:   [            9275]	Playback 2 (8): CEC version 1.4
DEBUG:   [            9275]	AllocateLogicalAddresses - device '0', type 'playback device', LA '8'
DEBUG:   [           10276]	command 'feature abort' timeout
DEBUG:   [           10776]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           10777]	<< e0
DEBUG:   [           11777]	command 'feature abort' timeout
DEBUG:   [           12278]	>> POLL not sent
DEBUG:   [           12278]	TV (0): device status changed into 'not present'
DEBUG:   [           12278]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           12278]	<< e0
DEBUG:   [           12278]	not sending command 'set osd name': destination device 'TV' marked as not present
DEBUG:   [           12278]	<< requesting power status of 'TV' (0)
DEBUG:   [           13279]	command 'feature abort' timeout
DEBUG:   [           13779]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           13779]	<< e0
DEBUG:   [           14779]	command 'feature abort' timeout
DEBUG:   [           15280]	>> POLL not sent
DEBUG:   [           15280]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [           15280]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           15280]	<< e0
DEBUG:   [           16281]	command 'feature abort' timeout
DEBUG:   [           16781]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [           16782]	<< e0
DEBUG:   [           17783]	command 'feature abort' timeout
DEBUG:   [           18284]	>> POLL not sent
DEBUG:   [           18285]	not sending command 'give device power status': destination device 'TV' marked as not present
ERROR:   [           18286]	failed to register a CEC client
unable to open the device on port RPI
DEBUG:   [           18288]	unregistering all CEC clients
NOTICE:  [           18289]	unregistering client: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Playback 2 (8) , base device: TV (0), HDMI port number: 1, physical address: f.f.f.f, git revision: libcec-6.0.2+1-76551ea, compiled on 2021-08-04 15:09:11 by pi@PItv on Linux 5.10.52-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi
DEBUG:   [           18290]	Playback 2 (8): power status changed from 'on' to 'unknown'
DEBUG:   [           18290]	Playback 2 (8): vendor = Unknown (000000)
DEBUG:   [           18291]	Playback 2 (8): CEC version unknown
DEBUG:   [           18291]	Playback 2 (8): device status changed into 'unknown'
Segmentation fault
```

popcornmix, 2021-08-05:
Could be this: https://github.com/Hexxeh/rpi-firmware/issues/267
The firmware should be appear in apt imminently.

sfoglia, 2021-08-05:
that's it now I've run the rpi-update and with libcec 4.0.7 is working again.

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234&p=1897156&hilit=cec#p1897156

`Linux PItv 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
`

closed 2021-08-05 by sfoglia