Pulse-Eight/libcec #579: Manjaro + Kodi + LG TV: CEC not working at all

naps62, 2021-09-07:
I have an LG C1 tv, a Raspberry Pi 4B on which I installed Manjaro and Kodi

On Kodi's settings, CEC does not even show as an option (in previous attempts that I had before settling on Manjaro, there was an option on Input -> Peripherals, even though I never got it to work)

I have installed `kodi-rpi`, which depends on `libcec-rpi`

`cec-client` gives me the following output (while connected to the TV):

```
naps62 in pi in ~
❯ cec-client -l
libCEC version: 6.0.2, git revision: <unknown>, compiled on 2021-03-27 18:45:45 by ray@pi4 on Linux 5.10.25-2-MANJARO-ARM (aarch64), features: P8_USB, DRM, P8_detect, randr, Linux
Found devices: NONE
```

I've tried a couple of different HDMI cables, tried all 4 of the tv's input ports, and tried disconnecting other HDMI devices (as I saw in another thread).
Kodi's debug logs have no mention of CEC at all

I'm running out of ideas, and need help

naps62, 2021-09-07:
Apparently changing the `PKGBUILD` of `libcec-rpi` to include `-DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib` seems to fix it on libcec's side (still testing Kodi)

can anyone confirm if this is an appropiate fix? in which case I guess the published pkgbuild is wrong (which I know is not this repo's responsibility)

popcornmix, 2021-09-08:
If you want to build cec to use the firmware cec driver then those defines are needed:
https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

When using the linux driver for cec (i.e. using `dtoverlay=vc4-kms-v3d`), those options aren't needed.

naps62, 2021-09-08:
I'm not sure what your first sentence means, to be honest. I'm not very familiar with how these libraries work

Are you saying the PKGBUILD from `libcec-rpi` is indeed wrong? (given that it's intended to target Raspberry Pi)

popcornmix, 2021-09-08:
I think these are questions for Manjaro.

I'm just saying you can build libcec for the firmware cec driver using instructions [here](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md).
Or you can build it using the linux cec framework (available when using `dtoverlay=vc4-kms-v3d`) by following instructions [here](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md).

I don't know know which Manjaro considers the default.

naps62, 2021-09-08:
manjaro uses no defaults. It just compiles from [this file](https://archlinuxarm.org/packages/armv7h/libcec-rpi/files/PKGBUILD), which defines the flags in the build() function explicitly

Indeed, I have `dtoverlay=vc4-fkms-v3d` on my system (added by kodi I believe?). Would this cause the compilation to run with `Raspberry pi support: NO`? (which is what was happening)
Or should it still say `YES` in that case?

PanopticaRising, 2021-09-09:
Thanks @naps62 for opening this thread, I recently set up a RPI4 with Manjaro and was banging my head against a wall, and thanks @popcornmix for the instructions.

I had compiled libcec myself to troubleshoot the issue, but I missed the key line:
```
-- Raspberry Pi support:                   yes
-- Linux support:                          yes
```

Even though I have `dtoverlay=vc4-fkms-v3d`, I still needed to add the RPI lib/include directories in `/opt/vc` for RPI support to show up. I recompiled with both,

```
cmake -DHAVE_LINUX_API=1  -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
```

And my cec-client _finally_ detects my LG TV!

Looking at the PKGBUILD that @naps62 posted, I'd wager the similar missing RPI_INCLUDE_* is the source of the problem.

PanopticaRising, 2021-09-10:
Correction, that PKGBUILD file had the `RPI_INCLUDE_` options included in a [previous commit](https://archlinuxarm.org/packages/armv7h/libcec-rpi/log?id=34489de78c1a5124f85651a26862f46aeece4e06). The maintainers removed it, referencing this in the commit: https://github.com/raspberrypi/linux/issues/4103

closed 2023-11-26 by naps62