Pulse-Eight/libcec #584: Incorrect CEC device on Raspberry Pi 4B

ReWiG, 2021-12-10:
I have installed the latest Raspberry Pi OS (Bullseye) on my Raspberry Pi 4B (which has only 2 HDMI ports). I am using the latest vc4-kms-v3d display driver.
`uname -a

Linux pikodi 5.10.83-v7l+ #1499 SMP Tue Dec 7 14:08:09 GMT 2021 armv7l GNU/Linux
`

I have libcec 6.0.2 library installed from the official repository of my operating system
For some reason, my system detects 3 CEC devices, 2 of them correspond to two HDMI ports, and the third is unknown, does not even have a COM port.
Here is the output of the `cec-client -l` command 

> libCEC version: 6.0.2, compiled on Linux-5.10.63-v7l+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
ERROR:   [            3543]     error opening serial port '': No such file or directory
ERROR:   [            3544]     could not open a connection (try 1)
ERROR:   [            6379]     error opening serial port '': No such file or directory
ERROR:   [            6379]     could not open a connection (try 2)
ERROR:   [            8269]     error opening serial port '': No such file or directory
ERROR:   [            8269]     could not open a connection (try 3)
ERROR:   [            9529]     error opening serial port '': No such file or directory
ERROR:   [            9529]     could not open a connection (try 4)
Found devices: 3
device:              1
com port:            /dev/cec0
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
device:              2
com port:            /dev/cec1
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
device:              3
com port:
vendor id:           0000
product id:          0000
firmware version:    65535

For comparison, here is the output of `cec-ctl --list-devices`

> vc4_hdmi (vc4):
        /dev/cec0
        /dev/cec1
Driver Info:
        Driver Name                : vc4_hdmi
        Adapter Name               : vc4
        Capabilities               : 0x0000011e
                Logical Addresses
                Transmit
                Passthrough
                Remote Control Support
                Connector Info
        Driver version             : 5.10.83
        Available Logical Addresses: 1
        DRM Connector Info         : card 1, connector 32
        Physical Address           : 2.0.0.0
        Logical Address Mask       : 0x0002
        CEC Version                : 1.4
        Vendor ID                  : 0x001582 (Pulse-Eight)
        OSD Name                   : ''
        Logical Addresses          : 1 (Allow Fallback to Unregistered)
          Logical Address          : 1 (Recording Device 1)
            Primary Device Type    : Record
            Logical Address Type   : Record

I would like device 3 to disappear, because it does not exist and does not work, but in my Kodi player it causes problems and the same errors are constantly written to the log file: 
`ERROR <general>: OpenConnection - could not opening a connection to the CEC adapter`
because it does not have a com port and Kodi cannot connect to it

popcornmix, 2021-12-10:
The third device is the firmware cec device (the other two are kernel).
You need the firmware device when running with legacy or fkms driver.
You need the kernel device when running with kms driver.

Ideally libcec would detect firmware driver is not available with kms, and would not report it.

ReWiG, 2021-12-10:
> The third device is the firmware cec device (the other two are kernel). You need the firmware device when running with legacy or fkms driver. You need the kernel device when running with kms driver.
> 
> Ideally libcec would detect firmware driver is not available with kms, and would not report it.

Is it possible to somehow disable it, because I am using vc4-kms-v3d and only kernel devices should be displayed for me

popcornmix, 2021-12-13:
The test for firmware driver being active is insufficient.
It checks for vchiq being available (which is for fkms and kms).
It should check for a cec firmware command (like set_passive) suceeding.
I'll see if I can get a PR (and get RPiOS repo updated).

Vodochnik, 2021-12-18:
WOW! I was wondering why CEC Pulse-Eight Adapter  complains about failing with exact the same error in log:
ERROR <general>: OpenConnection - could not opening a connection to the CEC adapter

Also 3 devices, last one cec adapter firmware v 65535. 

I have the same config (rpi 4, bullseye).

popcornmix, 2021-12-20:
If you are able to build libcec, then can you test https://github.com/popcornmix/libcec/tree/checkpassive ?
That should remove the unwanted firmware interface when using kms driver.
A prebuilt one is [here](https://drive.google.com/uc?id=15R_RbN9NQmWjoob1RA1w9_W1jzyDiTAU&export=download).
Replace the one in /lib/arm-linux-gnueabihf/libcec.so.6.0.2 (after taking a copy of it first).

ReWiG, 2021-12-21:
I tried using the file you provided and it doesn't seem to work. Here is the output of the "cec-client -l" command and I still see the same errors in the Kodi log 

> libCEC version: 6.0.2, git revision: libcec-6.0.2+2-6239ae1~dirty, compiled on 2021-12-20 15:07:14 by pi@pi4 on Linux 5.15.10-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
FindAdapter: iResult:0
FindAdapter: iResult:0
FindAdapter: iResult:0
ERROR:   [            3570]     error opening serial port '': No such file or directory
ERROR:   [            3570]     could not open a connection (try 1)
ERROR:   [            6407]     error opening serial port '': No such file or directory
ERROR:   [            6407]     could not open a connection (try 2)
ERROR:   [            8297]     error opening serial port '': No such file or directory
ERROR:   [            8297]     could not open a connection (try 3)
ERROR:   [            9557]     error opening serial port '': No such file or directory
ERROR:   [            9557]     could not open a connection (try 4)
Found devices: 3
device:              1
com port:            /dev/cec0
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
device:              2
com port:            /dev/cec1
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
device:              3
com port:
vendor id:           0000
product id:          0000
firmware version:    65535

I want to clarify that I used the file you provided and did not compile my own. If it needs to be done, let me know, I will try

popcornmix, 2021-12-21:
@ReWiG Can you try with this [url=https://drive.google.com/uc?id=1v7iJ4Cfm2dGiJkV8CF6BWSPGR510aCqU&export=download]firmware[/url] too?

ReWiG, 2021-12-21:
> @ReWiG Can you try with this [firmware](https://drive.google.com/uc?id=1v7iJ4Cfm2dGiJkV8CF6BWSPGR510aCqU&export=download) too?

To be honest, I have no idea what these files are and what to do with them. I'm not that good at this hardware. Can I have a little instruction, I'll try to do something?

ReWiG, 2021-12-22:
I replaced the firmware files and rebooted the device, here is the output from `cec-client -l `
> libCEC version: 6.0.2, git revision: libcec-6.0.2+2-6239ae1~dirty, compiled on 2021-12-20 15:07:14 by pi@pi4 on Linux 5.15.10-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
FindAdapter: iResult:0
FindAdapter: iResult:0
FindAdapter: iResult:8
Found devices: 2
device:              1
com port:            /dev/cec0
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
device:              2
com port:            /dev/cec1
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

It looks like replacing the firmware solves the problem, but what about my original firmware? And in case of a firmware update?

popcornmix, 2021-12-22:
The latest rpi-update firmware includes the fix. As will any future apt firmware updates, so you should be safe to update.

ReWiG, 2021-12-23:
Fine! I updated with rpi-update and the problem disappeared, now I have only 2 CEC devices. We will wait for the corrected version of the libcec library in the official Raspberry Pi OS repository.

closed 2021-12-23 by ReWiG

marvinme, 2021-12-24:
I had a same problem and fixed using the library and firmware.
By the way, it does not stop video playing, when I turn off TV.
It only stop video playing, when I turn on TV.
It is working ok with libreelec, which is libcec.so.4.0.7.
Is there any way to move to fix or move to 4.0.7?

popcornmix, 2021-12-24:
Have you chosen "stop video playback" as action for "when TV is switched off" in system/input/peripherals/cec settings?

marvinme, 2021-12-25:
Sure.
I am using the feature already with libreelec.
By the way, I set both "Action when switching to another source" & " When
the TV is switched off" to "Stop Playback".
Only "Action when switching to another source" works.
When I turn off TV, it does not stop playing.
When I turn on TV, it stops playing.
With Libreelec, both features works alright.

Regards
Marvin

marvinme, 2021-12-25:
I tried with osmc and it had no problem.
libCEC version: 6.0.2,  compiled on 2021-02-21 00:19:23 by root@compiler2 on Linux 4.19.0-9-amd64 (armv7l)

Vodochnik, 2021-12-25:
Big thanks to popcornmix, with new lib and rpi-update it works now. Lib was not in /lib/arm-linux-gnueabihf/ but in /usr/lib/arm-linux-gnueabihf/ (Armbian 11)

marvinme, 2021-12-26:
I tried again with osmc 20210808, 5.10.32-1-osmc.
Later versions 202111, 202112 of pi4 has the same problem.

popcornmix, 2021-12-26:
@marvinme did you update both cec lib and firmware? What does `vcgencmd version` report?

marvinme, 2021-12-27:
I updated both cec lib and firmware.
Dec 21 2021 13:25:37
Copyright (c) 2012 Broadcom
version 026fb7866b0cd7f9be209fcee3d57df69a95607c (tainted) (release) (start)
From my experience, it is ok with kodi 19.1, but has a problem with kodi 19.3.

fvisagie, 2021-12-31:
The updated library and firmware seem to work with my Kodi 19.3 on RPi 3B with fully updated RPiOS Bullseye, but there seems to be another problem.

Previously two adapters were shown although the 3B only has one. Now only 1 shows with both the KMS and fKMS drivers and both connect successfully.

However, as Kodi now exits it prints a long list of repeated notifications on the screen. The notifications only flash briefly before the terminal screen from which Kodi was launched is restored.

As far as I can make out, the notification text consists of the following continuously repeated 3 lines

> FindAdapter: iResult:0
> FindAdapter: iResult:0
> FindAdapter: iResult:8

The '8' could be a 'B', it is difficult to be sure. I took a video and frame-stepped through it, but the sudden changes in content and brightness create problems with exposure and focus.

Judging by the number of notifications produced during the time Kodi was running, I would guess the messages could be logged at the same 20s intervals as the OP's "OpenConnection" error message.

Nothing of this text shows up in the Kodi log even with all debugging including component-level debugging enabled. It also does not appear in the dmesg output. I've tried redirecting Kodi's stdout and stderr outputs to file but that did not capture the text either.

Let me know if any other information is needed.

popcornmix, 2022-01-01:
Yes that was debug. I'll try to get updated libcec into RPi repo without the debug.

fvisagie, 2022-01-02:
That would be great, thanks.

I guess that explains why the libcec you posted above is so large compared to the previous one?