Pulse-Eight/libcec #591: Assertion failure

igorkietrz, 2022-01-07:
I’m trying to turn on tv by a connected to rpi hdmi, but when running echo “scan” | cec-client -s -d 1 it gives me “assertion failure:./interface/vmcs_host/…/vc_vchi_dispmanx init():success == 0

popcornmix, 2022-01-07:
Is this RPiOS Bullseye or something older (`lsb_release -a`)?
What does "vcgencmd version" report?

igorkietrz, 2022-01-07:
Yes, It's rpios bullseye (openhab 3) and vcgencmd reports "d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start_cd)

igorkietrz, 2022-01-07:
> Is this RPiOS Bullseye or something older (`lsb_release -a`)? What does "vcgencmd version" report?

Yes, It's rpios bullseye (openhab 3) and vcgencmd reports "d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start_cd)

popcornmix, 2022-01-07:
It's because you are using the cutdown firmware. Try without the gpu_mem=16 in config.txt.

igorkietrz, 2022-01-07:
> It's because you are using the cutdown firmware. Try without the gpu_mem=16 in config.txt.

Thank you, It's working now

closed 2022-01-07 by igorkietrz