Pulse-Eight/libcec #597: cec-client fails to swtich HDMI inputs on raspbian

B0GDAN, 2022-03-06:
It appears that the cec-client has a serious bug on Raspbian GNU/Linux 11 (bullseye)

echo 'tx 4F:82:10:00' | cec-client -s -d 1
opening a connection to the CEC adapter...
ERROR:   [            4266]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22

Worked flawlessly on the previous version of raspbian. Any ideas, workarounds?

Thanks
Bogdan

B0GDAN, 2022-03-06:
echo "tx 4F:82:10:00" | cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              87]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              87]	CLinuxCECAdapterCommunication::Open - m_path=/dev/cec0 m_fd=4 bStartListening=1
DEBUG:   [              87]	CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_G_PHYS_ADDR - addr=2000
DEBUG:   [              87]	CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [              88]	CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=8000 num_log_addrs=1
NOTICE:  [              88]	connection opened
DEBUG:   [              88]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=8000 phys_addr=2000
DEBUG:   [              88]	processor thread started
DEBUG:   [              88]	<< Broadcast (F) -> TV (0): POLL
TRAFFIC: [              88]	<< f0
DEBUG:   [             394]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=1 addr=f0 opcode=ffffffff
DEBUG:   [             395]	>> POLL sent
DEBUG:   [             395]	TV (0): device status changed into 'present'
DEBUG:   [             395]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             395]	<< f0:8c
DEBUG:   [             462]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=2 addr=f0 opcode=8c
DEBUG:   [            1462]	expected response not received (87: device vendor id)
TRAFFIC: [            1462]	<< f0:8c
DEBUG:   [            1515]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=2 addr=f0 opcode=8c
DEBUG:   [            2516]	expected response not received (87: device vendor id)
DEBUG:   [            2516]	registering new CEC client - v6.0.2
DEBUG:   [            2516]	SetClientVersion - using client version '6.0.2'
NOTICE:  [            2516]	setting HDMI port to 1 on device TV (0)
DEBUG:   [            2517]	SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [            2517]	detecting logical address for type 'recording device'
DEBUG:   [            2517]	trying logical address 'Recorder 1'
DEBUG:   [            2517]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [            2517]	<< 11
DEBUG:   [            2579]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=11 opcode=ffffffff
TRAFFIC: [            2580]	<< 11
DEBUG:   [            2651]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=11 opcode=ffffffff
DEBUG:   [            2651]	>> POLL not sent
DEBUG:   [            2651]	using logical address 'Recorder 1'
DEBUG:   [            2651]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [            2651]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [            2652]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [            2652]	Recorder 1 (1): CEC version 1.4
DEBUG:   [            2652]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [            2652]	CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [            2652]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=2000
DEBUG:   [            2794]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0002 phys_addr=2000
DEBUG:   [            2794]	changing physical address to 2000
DEBUG:   [            2794]	Recorder 1 (1): physical address changed from ffff to 2000
DEBUG:   [            2794]	CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0002 num_log_addrs=1
DEBUG:   [            2795]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [            2795]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [            2795]	using provided physical address 2000
NOTICE:  [            2795]	CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 0, logical address(es) = Recorder 1 (1) , physical address: 2.0.0.0, compiled on Linux-5.10.63-v8+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
DEBUG:   [            2795]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            2795]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            3351]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=11 addr=10 opcode=47
DEBUG:   [            3351]	<< requesting power status of 'TV' (0)
TRAFFIC: [            3351]	<< 10:8f
DEBUG:   [            3418]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=2 addr=10 opcode=8f
DEBUG:   [            3502]	CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=2 addr=01 opcode=8f
TRAFFIC: [            3502]	>> 01:8f
DEBUG:   [            3502]	<< Recorder 1 (1) -> TV (0): on
TRAFFIC: [            3502]	<< 10:90:00
DEBUG:   [            3502]	>> TV (0) -> Recorder 1 (1): give device power status (8F)
DEBUG:   [            3589]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=3 addr=10 opcode=90
DEBUG:   [            3662]	CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=2 addr=01 opcode=46
[... cut ...]
DEBUG:   [            4042]	unregistering all CEC clients
DEBUG:   [            4042]	CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [            4047]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=2000
DEBUG:   [            5053]	CLinuxCECAdapterCommunication::Process - stopped - m_path=/dev/cec0 m_fd=4
DEBUG:   [            5053]	CLinuxCECAdapterCommunication::Close - m_path=/dev/cec0 m_fd=4

B0GDAN, 2022-03-16:
pi@xx:~ $ vcgencmd version
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
pi@xx:~ $ uname -a
Linux xx 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux

I will try your recommendation regarding dtoverlay

Jibun-no-Kage, 2022-05-29:
Anyone know WHY this is happening?

darton, 2022-11-21:
echo 'on 0.0.0.0' | cec-client -s -d 1 
opening a connection to the CEC adapter...
Segmentation fault

It takes quite a long time and ends with an error, but it works

#dtoverlay=vc4-kms-v3d
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

----------------------
Oct 26 2022 11:09:05 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux

jdgorman58, 2023-03-20:
> Post output of: vcgencmd version uname -a
> 
> and contents of config.txt. Does it work if you change `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` in config.txt (and reboot).

Had the same issues until I found this thread.  The change suggested work flawlessly..  Thank you!

iwl, 2024-02-17:
I still get that error:   tx 1f:82:11:00
ERROR:   [           13518]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22

milux, 2024-11-18:
> Does it work if you change `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` in config.txt (and reboot).

This can have serious performance impact even when it works.
### Our Pi stopped booting its UI after this change!

For us, the issue lied with the command itself.
We realized the mistake after a long hour of frustrating debugging attempts when we executed `echo 'scan' | cec-client -s -d 1` and saw that `Recorder 1` (Pulse Eight CECTester) had changed its address from `2.0.0.0` to `1.0.0.0`.
Turned out that the "source address" contained in the command at byte 1 had to be changed due to the upgrade:

`echo 'tx 2F:82:10:00' | cec-client -s -d 1` output (not working):
```
opening a connection to the CEC adapter...
ERROR:   [            2062]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
```

`echo 'tx 1F:82:10:00' | cec-client -s -d 1` output (**working!**):
```
opening a connection to the CEC adapter...
```

So, it might be as easy as trying a slightly different address...

Jibun-no-Kage, 2025-02-26:
Cannot test with CEC 7.0... not available...
```
Get:1 http://deb.debian.org/debian bookworm/main arm64 libp8-platform2 arm64 2.1.0.1+dfsg1-4 [20.9 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libcec6 arm64 6.0.2-5+rpt2 [188 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 cec-utils arm64 6.0.2-5+rpt2 [30.0 kB]
```

Jibun-no-Kage, 2025-02-26:
Seems that original scan error fixed overall with Debian 12 based PiOS release?
```
# echo 'scan' | cec-client -d 1
log level set to 1
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect:
 path:     /sys/devices/platform/soc/3f902000.hdmi/cec0
 com port: /dev/cec0

opening a connection to the CEC adapter...
waiting for input
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```
But the active source issue remains.

pka4916, 2025-02-27:
wtf  do i get hundreds of e-mails from all the posts...

remove me...  for gods sake....

opdenkamp, 2025-02-27:
stop posting the exact same comment in every ticket @Jibun-no-Kage. removing your comments here.

@pka4916 you have subscribed yourself to updates from this repository and a bot just closed a lot of old tickets in here. if you don't want to receive those emails, then you have to unsubscribe.

pka4916, 2025-02-27:
I can't seem to delete them.  And I never subscribed to it

opdenkamp, 2025-02-28:
@pka4916 if you're getting those emails, then you're watching this repository. check the button watch/unwatch on https://github.com/Pulse-Eight/libcec to control this. I suspect that you've clicked watch when you starred this repos. I can't control this for you, you have to change this yourself to get rid of this email spam.

opdenkamp, 2025-02-28:
Is this still an issue when using libCEC 7.0.0?

closed 2025-05-30 by github-actions[bot]