Pulse-Eight/libcec #600: User Control commands (44) passed to Apple TV produces duplicate actions

HVR88, 2022-05-01:
Menu navigation and volume commands (for example) produce duplicate actions when passed to Apple TV (2021) devices.

**Example sending user control RIGHT to Apple TV as device 4:**
echo tx "F4:44:04" | cec-client -s -d 1

**Expected behavior**
Menu selection moves ONE position to the right

**Result**
Menu selection moves TWO positions

- The same behaviour can be seen with any menu direction as well as volume up/down. 
- No errors are reported from cec-client when this happens.
- Must use an unregistered source device  (f, d, e etc.)or there is no action at all
- Same commands passed to NVIDIA SHIELD (2017) works as expected (also can use valid source devices, but not TV(0))

malard, 2022-05-01:
This would be a bug with the Apple TV then, but sending a broadcast control like this is not supported as per the spec

Change it to “04:44:04” and see what happens.

HVR88, 2022-05-01:
Using source 0 (TV) does not work for the devices I have.  

No result with Apple TV (Playback 1), NVIDIA SHIELD (Playback 3), nor Marantz AV7703 (Audio)

malard, 2022-05-01:
They need to be the active source first

HVR88, 2022-05-02:
While each device is being used, it is the active device - as reported by "ad {addr}"

Attempting to manually specify said device as active using "spl {addr}" or "sp {addr}" produces an error

"only the TV is allowed to send CEC_OPCODE_SET_STREAM_PATH"

Here's some output

```
ad B
DEBUG:   [          393059]	<< Recorder 1 (1) -> Recorder 2 (2): POLL
TRAFFIC: [          393059]	<< 12
TRAFFIC: [          393150]	<< 12
DEBUG:   [          393240]	>> POLL not sent
DEBUG:   [          393240]	Recorder 2 (2): device status changed into 'not present'
DEBUG:   [          393240]	<< Recorder 1 (1) -> Tuner 1 (3): POLL
TRAFFIC: [          393241]	<< 13
TRAFFIC: [          393331]	<< 13
DEBUG:   [          393421]	>> POLL not sent
DEBUG:   [          393421]	Tuner 1 (3): device status changed into 'not present'
DEBUG:   [          393421]	<< Recorder 1 (1) -> Playback 1 (4): POLL
TRAFFIC: [          393421]	<< 14
DEBUG:   [          393452]	>> POLL sent
DEBUG:   [          393452]	Playback 1 (4): device status changed into 'present'
DEBUG:   [          393452]	<< Recorder 1 (1) -> Audio (5): POLL
TRAFFIC: [          393452]	<< 15
DEBUG:   [          393513]	>> POLL sent
DEBUG:   [          393513]	Audio (5): device status changed into 'present'
DEBUG:   [          393513]	<< Recorder 1 (1) -> Tuner 2 (6): POLL
TRAFFIC: [          393513]	<< 16
TRAFFIC: [          393604]	<< 16
DEBUG:   [          393694]	>> POLL not sent
DEBUG:   [          393694]	Tuner 2 (6): device status changed into 'not present'
DEBUG:   [          393695]	<< Recorder 1 (1) -> Tuner 3 (7): POLL
TRAFFIC: [          393695]	<< 17
TRAFFIC: [          393785]	<< 17
DEBUG:   [          393875]	>> POLL not sent
DEBUG:   [          393875]	Tuner 3 (7): device status changed into 'not present'
DEBUG:   [          393875]	<< Recorder 1 (1) -> Playback 2 (8): POLL
TRAFFIC: [          393875]	<< 18
TRAFFIC: [          393966]	<< 18
DEBUG:   [          394056]	>> POLL not sent
DEBUG:   [          394056]	Playback 2 (8): device status changed into 'not present'
DEBUG:   [          394056]	<< Recorder 1 (1) -> Recorder 3 (9): POLL
TRAFFIC: [          394056]	<< 19
TRAFFIC: [          394146]	<< 19
DEBUG:   [          394237]	>> POLL not sent
DEBUG:   [          394237]	Recorder 3 (9): device status changed into 'not present'
DEBUG:   [          394237]	<< Recorder 1 (1) -> Tuner 4 (A): POLL
TRAFFIC: [          394237]	<< 1a
TRAFFIC: [          394328]	<< 1a
DEBUG:   [          394418]	>> POLL not sent
DEBUG:   [          394418]	Tuner 4 (A): device status changed into 'not present'
DEBUG:   [          394418]	<< Recorder 1 (1) -> Reserved 1 (C): POLL
TRAFFIC: [          394418]	<< 1c
TRAFFIC: [          394509]	<< 1c
DEBUG:   [          394599]	>> POLL not sent
DEBUG:   [          394599]	Reserved 1 (C): device status changed into 'not present'
DEBUG:   [          394599]	<< Recorder 1 (1) -> Reserved 2 (D): POLL
TRAFFIC: [          394600]	<< 1d
TRAFFIC: [          394690]	<< 1d
DEBUG:   [          394780]	>> POLL not sent
DEBUG:   [          394780]	Reserved 2 (D): device status changed into 'not present'
DEBUG:   [          394780]	<< Recorder 1 (1) -> Free use (E): POLL
TRAFFIC: [          394781]	<< 1e
TRAFFIC: [          394871]	<< 1e
DEBUG:   [          394961]	>> POLL not sent
DEBUG:   [          394961]	Free use (E): device status changed into 'not present'
logical address B is active
tx 0B:44:41
waiting for input
DEBUG:   [          433315]	initiator 'TV' is not supported by the CEC adapter
```

HVR88, 2022-05-02:
And the debug output from the command-line:

```
echo tx "0B:44:41" | cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              94]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              94]	logical address changed to Free use (e)
DEBUG:   [              94]	Open - vc_cec initialised
NOTICE:  [              94]	connection opened
DEBUG:   [              95]	processor thread started
DEBUG:   [              95]	<< Broadcast (F) -> TV (0): POLL
DEBUG:   [              95]	initiator 'Broadcast' is not supported by the CEC adapter. using 'Free use' instead
TRAFFIC: [              95]	<< e0
DEBUG:   [             155]	>> POLL sent
DEBUG:   [             155]	TV (0): device status changed into 'present'
DEBUG:   [             155]	<< requesting vendor ID of 'TV' (0)
TRAFFIC: [             155]	<< e0:8c
TRAFFIC: [             351]	>> 0f:87:00:00:f0
DEBUG:   [             351]	>> TV (0) -> Broadcast (F): device vendor id (87)
DEBUG:   [             352]	TV (0): vendor = Samsung (0000f0)
DEBUG:   [             352]	expected response received (87: device vendor id)
DEBUG:   [             352]	replacing the command handler for device 'TV' (0)
DEBUG:   [             352]	registering new CEC client - v6.0.2
DEBUG:   [             352]	SetClientVersion - using client version '6.0.2'
NOTICE:  [             352]	setting HDMI port to 1 on device TV (0)
DEBUG:   [             352]	SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [             353]	detecting logical address for type 'recording device'
DEBUG:   [             353]	trying logical address 'Recorder 1'
DEBUG:   [             353]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             353]	<< 11
TRAFFIC: [             623]	<< 11
DEBUG:   [             894]	>> POLL not sent
DEBUG:   [             894]	using logical address 'Recorder 1'
DEBUG:   [             894]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             895]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             895]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             895]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             895]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             896]	logical address changed to Recorder 1 (1)
DEBUG:   [             896]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             896]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             896]	GetPhysicalAddress - physical address = 1000
DEBUG:   [             897]	using auto-detected physical address 1000
DEBUG:   [             897]	Recorder 1 (1): physical address changed from ffff to 1000
DEBUG:   [             897]	<< Recorder 1 (1) -> broadcast (F): physical address 1000
TRAFFIC: [             897]	<< 1f:84:10:00:01
NOTICE:  [            1048]	CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, compiled on Linux-5.10.63-v8+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
DEBUG:   [            1048]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [            1048]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [            1349]	<< requesting power status of 'TV' (0)
TRAFFIC: [            1349]	<< 10:8f
TRAFFIC: [            1493]	>> 01:90:00
DEBUG:   [            1493]	>> TV (0) -> Recorder 1 (1): report power status (90)
DEBUG:   [            1493]	TV (0): power status changed from 'unknown' to 'on'
DEBUG:   [            1493]	expected response received (90: report power status)
DEBUG:   [            1494]	initiator 'TV' is not supported by the CEC adapter
DEBUG:   [            1494]	unregistering all CEC clients
NOTICE:  [            1494]	unregistering client: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 1, logical address(es) = Recorder 1 (1) , physical address: 1.0.0.0, compiled on Linux-5.10.63-v8+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
DEBUG:   [            1495]	Recorder 1 (1): power status changed from 'on' to 'unknown'
DEBUG:   [            1495]	Recorder 1 (1): vendor = Unknown (000000)
DEBUG:   [            1495]	Recorder 1 (1): CEC version unknown
DEBUG:   [            1495]	Recorder 1 (1): osd name set to 'Recorder 1'
DEBUG:   [            1495]	Recorder 1 (1): device status changed into 'unknown'
DEBUG:   [            1496]	unregistering all CEC clients
DEBUG:   [            2095]	UnregisterLogicalAddress - releasing previous logical address
DEBUG:   [            2096]	logical address changed to Broadcast (f)
```

HVR88, 2022-05-02:
Using 1 (the address of the Rpi/CEC adapter) works (the same as F) when testing with Shield or Apple TV as sink/destination.

Still produces double actions on Apple TV for navigation and volume and on Shield normal/single actions, but oddly, inside the Amazon Prime app, navigation actions (not volume) are doubled just like on Apple TV.

HVR88, 2022-05-06:
I received the CEC adapter today and unlike the RPi, it works with the 0 address and it can scan CEC commands as they're being sent by other devices and remotes.

I'm quite confident the reason for the double actions in both Apple TV and the Amazon Prime app on Android TV are due to repeat rate of those implementations and likely the slow to terminate response from the CEC library.

When using the TV remote to move a direction by a single press, the 45 (button up) signal is immediately sent and each platform responds precisely one time.  When sending a raw command using cec-client, I can only send the single command and can't modify repeat rate or add a button up. 

This should be able to be remedied with some timing adjustments.

HVR88, 2022-05-14:
@malard now that I have the Pulse-Eight adapter, I'm happy to provide more debugging/testing. 

I'd really like to get this resolved as I suspect it affects more platforms than just the two I've found so far. Being able to send tight sequence of commands would go a long way to figuring this out. Example:  direction->direction->button-up

HVR88, 2025-02-26:
Still an issue

opdenkamp, 2025-02-27:
this is what's expected to happen really when you only send opcode 0x44 without 0x45. the receiving device will treat this as key down without key up and will time out the key up event eventually. this command without 0x45 will result in even more key presses depending on the device that's being addressed.

sending raw commands with the `tx` and `txn` command are not supported because it just injects raw CEC data on the bus without invoking any high level code. cec-client allows you to do this (its main purpose is a debugging tool) but what it does isn't supported.

but what you're asking for here is a higher level command to send key up + key down with cec-client. I'll change the title to match this

closed 2025-05-29 by github-actions[bot]