Pulse-Eight/libcec #618: Source is not selected when starting device

Nonostanovich, 2022-09-30:
Hi,

Tv starts when I launch the PC, but source is not changed to HDMI 1. It stays to the last used input.

On Windows 10, Pulse-Height USB CEC adpater (firmware v12), and libcec 6.0.2 installed

Configuration :
Connected to HDMI Device : Philips TV
HDMI port of the TV : HDMI 1
Override physical addresse : unchecked (1000)
Power Control :
Power on the TV When the PC starts  : checked

Device Information :
Device TV
Logical address 0
Physical address 0
Device present yes
Active source no
Vendor Philips
CEC version 1.4
Power status on
OSD name TV
Menu language fre

Did I made something wrong ?

Nonostanovich, 2022-09-30:
Has I am not the only user, session don't start until I have to login.
Does CEC Tray have to be start in order to active the right input ?

pka4916, 2023-10-15:
i am having the same problem..   my default is HDMI 3, and it keeps going back to 1.
the settings are not saved either..

The batch files don't help either..