Pulse-Eight/libcec #621: Option to prevent/ignore powering down PC

pzich, 2022-10-14:
I would like to be able to control powering my TV on and off from my PC.

In order to do that from my LG tv, I have to enable "Auto Power Sync" which enables the following behavior:
- When the device is turned on/off, your TV will automatically turn on/off
- When you TV is turned off, the device will automatically turn off.

I would like the first behavior, but when I turn my TV off I would like my PC to remain on instead of going into hibernation.

---

Additionally, with this setting on it seems that frequently when I turn my TV off, the PC will also turn off, but turn back on shortly after and turn the TV back on.

pzich, 2024-01-17:
@malard Any chance we could get this configuration/customization added? 🙏 I really don't want anything putting my PC to sleep as it has apps/services running that I want to remain active

malard, 2024-01-17:
If you have already written the code then you can compile it yourself, if you are asking us to write code  sorry we don't have the spare capacity at the moment

pzich, 2024-01-19:
Thanks for replying. I've followed the setup steps and attempted to compile it myself, but keep running into compilation errors.

I'm not very familiar with development on the windows side, so I'm guessing it's a basic issue, but I seem to be stuck.

---

Thank you for making this product! I've been using one for 3 years now and it's been life changing for my TV->PC usage.

closed 2025-02-27 by github-actions[bot]