Pulse-Eight/libcec #623: Wake PC on CEC input

pzich, 2022-10-14:
If my PC enters the screensaver, I am unable to use CEC input (e.g. play pause, arrows) to wake it back up. I instead have to use keyboard/mouse/VNC to wake the device first.

Is it possible to make these inputs wake the PC? I thought this functionality had been added in 6.0.0 but it doesn't seem to be working for me.

satmandu, 2023-03-14:
I would also like this to happen. 

Thinking I might be able to use something like https://github.com/bramp/libcec-daemon with ydotool to move the mouse on HDMI activation.

pzich, 2024-01-16:
@malard @opdenkamp could we get this functionality added?

malard, 2024-01-16:
I will see if we can make time before end of the month!

pzich, 2024-04-08:
Any chance you can take a look this month?

Disgruntled-Duck, 2024-05-02:
I have this same issue, I have to use VNC or IR to wake up my PC.

satmandu, 2024-05-02:
So I'm wondering if libcec needs a hook into dbus or something to engage wake from sleep.

Libcec is happy to report data from the connected adapter, but I think dbus or something normally listens for certain devices... So the question is maybe, should libcec have a mode where it emulates a simple keyboard so that when a button on a remote is pressed, libcec reports it as a keyboard press on an attached keyboard.

satmandu, 2024-05-02:
Basically it would be nice for the adapter to have a mode where it can do what this hardware does:

https://hackaday.io/project/168696-cec2usb

Disgruntled-Duck, 2024-05-03:
I am sure this is a super niche case but I figured I would throw it out there anyway for anybody it may help. I use home assistant and integrated my TV into home assistant (you could also probably do an ICMP check). I also setup WoL in home assistant and on my mini PC. I then setup an automation that when the TV is detected on, send WoL to the PC. It has worked perfectly the last day.

closed 2025-02-27 by github-actions[bot]

pzich, 2025-02-27:
Please add this feature. It would be very useful