Pulse-Eight/libcec #629: Python cec Errors

mtullo27, 2023-02-27:
When running the sample code located [here](https://github.com/Pulse-Eight/libcec/tree/master/src/pyCecClient), I get the following errors.

```
<module 'cec' from '/home/mtullo/.local/lib/python3.9/site-packages/cec.cpython-39-arm-linux-gnueabihf.so'>
Traceback (most recent call last):
  File "/home/mtullo/Desktop/tulloVision.py", line 38, in <module>
    class pyCecClient:
  File "/home/mtullo/Desktop/tulloVision.py", line 39, in pyCecClient
    cecconfig = cec.libcec_configuration()
AttributeError: module 'cec' has no attribute 'libcec_configuration'
```

drolevar, 2023-04-07:
Most probably you have python-cec installed with pip (this lib https://github.com/trainman419/python-cec). Both this lib and python-cec unfortunately have "cec" as module name, so they conflict. I suggest you to run "pip uninstall cec".

albfan, 2024-05-03:
After ensure libcec python is the only package:

```
$ python
Python 3.11.7 (main, Dec 18 2023, 00:00:00) [GCC 13.2.1 20231011 (Red Hat 13.2.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> import sys
>>> import pprint
>>> 
>>> pprint.pprint(sys.path)
['',
 '/usr/lib64/python311.zip',
 '/usr/lib64/python3.11',
 '/usr/lib64/python3.11/lib-dynload',
 '/home/alberto/.local/lib/python3.11/site-packages',
 '/usr/local/lib64/python3.11/site-packages',
 '/usr/local/lib/python3.11/site-packages',
 '/usr/lib64/python3.11/site-packages',
 '/usr/lib/python3.11/site-packages',
 '/usr/lib/python3.11/site-packages/screenkey-1.5-py3.11.egg']
>>> 
$ fd cec.py ~/.local/
#no result

$ fd cec.py /usr
/usr/lib64/python3.11/site-packages/cec/cec.py

$ head /usr/lib64/python3.11/site-packages/cec/cec.py
# This file was automatically generated by SWIG (https://www.swig.org).
# Version 4.1.1
#
# Do not make changes to this file unless you know what you are doing - modify
# the SWIG interface file instead.

from sys import version_info as _swig_python_version_info
# Import the low-level C/C++ module
if __package__ or "." in __name__:
    from . import _cec
```

I conclude the script is wrong:

```
diff --git i/src/pyCecClient/pyCecClient.py w/src/pyCecClient/pyCecClient.py
index 8b18d88..f7e16e9 100755
--- i/src/pyCecClient/pyCecClient.py
+++ w/src/pyCecClient/pyCecClient.py
@@ -32,8 +32,7 @@
 #     http://www.pulse-eight.com/
 #     http://www.pulse-eight.net/
 
-import cec
-print(cec)
+from cec import cec
 
 class pyCecClient:
   cecconfig = cec.libcec_configuration()
```

For me it needs to use sudo but now it works

```
$ sudo ./pyCecClient.py 
libCEC version 6.0.2 loaded:  compiled on 2023-01-19 00:00:00 by mockbuild on Linux 6.1.5-200.fc37.x86_64 (x86_64), features: P8_USB, DRM, P8_detect, randr, Linux
NOTICE:  [1307]     connection opened
NOTICE:  [2241]     connection opened
found a CEC adapter:
port:     /dev/ttyACM0
vendor:   0x2548
product:  0x1002
found a CEC adapter:
port:     Linux
vendor:   0x0
product:  0x0
NOTICE:  [2243]     connection opened
TRAFFIC: [2243]     << f0
^CTRAFFIC: [4597]     << f0
NOTICE:  [4673]     setting HDMI port to 1 on device TV (0)
TRAFFIC: [4673]     << f0
TRAFFIC: [4750]     << f0
TRAFFIC: [4827]     << 11
```

opdenkamp, 2025-02-26:
fixed in libCEC 7.0.0

closed 2025-02-26 by opdenkamp