Pulse-Eight/libcec #631: RPI4: cannot connect to /dev/cec1

alien999999999, 2023-03-01:
Executing cec-client makes it connect to /dev/cec0 (according to the debug lines)

I don't know how to connect to /dev/cec1 , i tried:

`cec-client /dev/cec1` but that didn't work, something about cannot lock serial, but then
`cec-client /dev/cec0` did not work either with the same error...

How can i make cec-client connect to /dev/cec1 ? cause the adapter is in the 2nd port

`cec-client -l` shows only one device:
```
Found devices: 1

device:              1
com port:            Linux
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

```
cec-ctl does show that the /dev/cec1 is the correct one, it shows a TV:
```
[root@rpimedia ~]# cec-ctl -S
Driver Info:
        Driver Name                : vc4_hdmi
        Adapter Name               : vc4
        Capabilities               : 0x0000011e
                Logical Addresses
                Transmit
                Passthrough
                Remote Control Support
                Connector Info
        Driver version             : 5.15.88
        Available Logical Addresses: 1
        DRM Connector Info         : card 0, connector 32
        Physical Address           : f.f.f.f
        Logical Address Mask       : 0x0000
        CEC Version                : 2.0
        OSD Name                   : ''
        Logical Addresses          : 0 (Allow Fallback to Unregistered)
[root@rpimedia ~]# cec-ctl -d 1 -S
Driver Info:
        Driver Name                : vc4_hdmi
        Adapter Name               : vc4
        Capabilities               : 0x0000011e
                Logical Addresses
                Transmit
                Passthrough
                Remote Control Support
                Connector Info
        Driver version             : 5.15.88
        Available Logical Addresses: 1
        DRM Connector Info         : card 0, connector 40
        Physical Address           : f.f.f.f
        Logical Address Mask       : 0x0000
        CEC Version                : 2.0
        Vendor ID                  : 0x000c03 (HDMI)
        OSD Name                   : 'Playback'
        Logical Addresses          : 1 (Allow RC Passthrough)

          Logical Address          : Not Allocated
            Primary Device Type    : Playback
            Logical Address Type   : Playback
            All Device Types       : Playback
            RC TV Profile          : None
            Device Features        :
                None

        System Information for device 0 (TV) from device 15 (Unregistered):
                CEC Version                : 
                Physical Address           : 
                Vendor ID                  : 
                OSD Name                   : 

        Topology:

```

alien999999999, 2023-03-02:
So, this is a mainline kernel and libcec is not built with rpi support, because of missing bcm_host and vchiq_arm libraries. I'm assuming the one entry is not /dev/cec0 or /dev/cec1 but the "linux" one? but cec-ctl can show me some TV stuff on the /dev/cec1 interface?

6by9, 2023-03-17:
Duplicates #538

Libcec is hard coded to use /dev/cec0 based on https://github.com/Pulse-Eight/libcec/blob/master/include/cectypes.h#L287

closed 2025-02-27 by github-actions[bot]