Pulse-Eight/libcec #632: Searching for way to query which TV input is active?

Jibun-no-Kage, 2023-03-21:
Searching for way to query which TV input is active?  I can't seem to find a 'query' command to determine the active input?  Is such even supported in CEC command list?  I tried searching CEC matic, but did not seem to find anything?

I know you can query power state via 'pow X' where X is the device address or device #, so looking for the same idea to determine the active input.

Jibun-no-Kage, 2025-02-26:
Seems that original scan error fixed overall with Debian 12 based PiOS release?
```
# echo 'scan' | cec-client -d 1
log level set to 1
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect:
 path:     /sys/devices/platform/soc/3f902000.hdmi/cec0
 com port: /dev/cec0

opening a connection to the CEC adapter...
waiting for input
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```
But the active source issue remains.  And the ability to query, discover, devices remains missing!

pka4916, 2025-02-27:
wtf  do i get hundreds of e-mails from all the posts...

remove me...  for gods sake....

opdenkamp, 2025-02-27:
> Cannot test with CEC 7.0... not available...
> 
> ```
> Get:1 http://deb.debian.org/debian bookworm/main arm64 libp8-platform2 arm64 2.1.0.1+dfsg1-4 [20.9 kB]
> Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libcec6 arm64 6.0.2-5+rpt2 [188 kB]
> Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 cec-utils arm64 6.0.2-5+rpt2 [30.0 kB]
> ```

Debian doesn't instantly include all new updates, that's more something for Arch

> But the active source issue remains. And the ability to query, discover, devices remains missing!
There's no TV connected in your setup, at least not to that HDMI output that libCEC is accessing here, which is mandatory according to the spec. So this is expected.

> wtf do i get hundreds of e-mails from all the posts...
You must have subscribed yourself to updates from this repository, this is not something that I control.

opdenkamp, 2025-02-27:
see https://github.com/Pulse-Eight/libcec/issues/441#issuecomment-2688427145

closed 2025-02-27 by opdenkamp