Pulse-Eight/libcec #637: How and when to use "audio_get_status"

chincheta0815, 2023-07-13:
I want to read out the audio volume and status from my audio system.
This does work only once, after I set the system into the active audio mode, I never again get any audio status...

Are there any restrictions or thing I have to consider???
What is the purpose of that command: using it once or periodically?

satmandu, 2023-10-13:
Are you planning on submitting a PR for the changes at https://github.com/chincheta0815/libcec/tree/fixes_and_additions ?

Did you figure out a way to get your audio volume and status?

Looking at the traffic in `cec-client` when I hit volume up on my remote I see volume go from 0:
```
TRAFFIC: [         1164987]     >> 05:89:10:75:00
TRAFFIC: [         1170579]     >> 05:89:10:75:01
```
All the way up to 100:
```
TRAFFIC: [         1235454]     >> 05:89:10:75:48
TRAFFIC: [         1235605]     >> 05:89:10:75:4a
TRAFFIC: [         1235708]     >> 05:89:10:75:4c
TRAFFIC: [         1235963]     >> 05:89:10:75:4e
TRAFFIC: [         1236163]     >> 05:89:10:75:50
TRAFFIC: [         1236363]     >> 05:89:10:75:52
TRAFFIC: [         1236564]     >> 05:89:10:75:54
TRAFFIC: [         1236763]     >> 05:89:10:75:56
TRAFFIC: [         1236963]     >> 05:89:10:75:58
TRAFFIC: [         1237163]     >> 05:89:10:75:5a
TRAFFIC: [         1237364]     >> 05:89:10:75:5c
TRAFFIC: [         1237563]     >> 05:89:10:75:5e
TRAFFIC: [         1237764]     >> 05:89:10:75:60
TRAFFIC: [         1237963]     >> 05:89:10:75:62
TRAFFIC: [         1238163]     >> 05:89:10:75:64
```

But sending volup or voldown using cec-client doesn't appear to change this.

(I have a Panasonic tv, connected to a Panasonic soundbar over CEC, and the soundbar is the primary volume source.)

audio_get_status also doesn't do anything for me....

chincheta0815, 2023-10-14:
Well, actually I do not know if I should submit a PR as there seems nobody actually to maintain libcec anymore...
I collected some PR's, bugfixes and additions that I understood and that seemed reasonable and helpful to me.
So far my branch is really subject to change at the moment... 

Anyways, I got the get_audio_status issue solved (with my Teufel Soundbar).
First of all the device (e.g. amp) needs to be set to "SYSTEM AUDIO MODE". Afterwards, you can change the volume via volup, voldown and mute via CEC.

To get a feedback there seems to be a bug/missing feature in the original code... I did a workaround/fix to get the audio status working. With the original code I only got the very first audio_status after setting the CEC device into AUDIO MODE. 

Only thing: My Amp does not show any mute state due to a seemingly lacking implementation of the mute mask...

malard, 2023-10-14:
Hi, we do maintain it, however we've been extremely busy at Pulse-Eight, I can get someone to review everything being requested soon

chincheta0815, 2023-10-14:
Hej Martin,

thanks a lot for the reply.
This was not ment to be offending. So sorry if these was a misunderstanding.

Actually I would really like to have some one to review the things.
As you are busy I would probably say: Not all.

So when I am done I would put it into parts or at least clean up a little bit.

Especially the audio_status seems to have probably a missing feature/bug.
Most other things I did are mainly to make the C version working…

Is that idea okay for you?

Best regards, Thanks a lot for the reply and Sorry for any misunderstandings...

Rouven

chincheta0815, 2023-10-19:
@malard I am coming back to our correspondence concerning reviewing all the requests by someone on your side. I am right now cleaning up a little bit and fell over a question I had when fixing some things. This question is related to outputting ther audio_status of a CEC device. (get_audio_status in cecc.h). In your last version in the master branch this outputs only the first audio status - every status afterwards else is not updated. 

I Managed to update that, but: How do you wish to have it updated/how is it best?
1. Internally in libcec so that the user should not have to cope with the correct timeouts. After a certain timeout (250ms) the function will update the status itself - the user just calls the function to her liking (e.g. https://github.com/chincheta0815/libcec/blob/90f67e9435d3f0e726340658e67ac4957f672b96/src/libcec/devices/CECAudioSystem.cpp#L178) 
or 
2. give an option to the user like "update" to get the latest values - here the user has to cope with the timing. (e.g. call GetAudioStatus with bUpdate = true from the user
3. interface...)
The original code does not show me the intention.

Both works, but is different in use. The solution should integrate to the rest of libcec and it's philiosophy...

chincheta0815, 2025-02-26:
I just send a pull request concerning these issue... Okay. the old one with conflicting CMake stuff. But if all rest still holds I would clean up...

MP-Munich, 2025-03-15:
Same for me: The correct volume level is returned but when I change the volume still the previous volume level is returned. When I reconnect to the CEC adapter, the new correct volume level is returned.

closed 2025-06-14 by github-actions[bot]