Pulse-Eight/libcec #639: USB adapter stopped working

Littletycoon, 2023-07-29:
After years, the USB-CEC adapter stopped working overnight, just sopped working. when running cec client I get following

WARNING: [            2550]     HIGH_ERROR line:218 time:1363
DEBUG:   [            2566]     command 'PING' was not acked by the controller
ERROR:   [            2566]     the adapter did not respond correctly to a ping (try 2)
with more tries, followed by:
WARNING: [           49312]     HIGH_ERROR line:218 time:503
DEBUG:   [           49327]     command 'GET_BUILDDATE' was not acked by the controller
and then some variations with also no acknowledgements from controller.

and then finally ending with below:
WARNING: [           55305]     HIGH_ERROR line:218 time:503
DEBUG:   [           75894]     CEC::CCECClient::SetClientVersion - using client version '4.0.7'
DEBUG:   [           75894]     CEC::CCECClient::SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [           75894]     detecting logical address for type 'recording device'
DEBUG:   [           75894]     trying logical address 'Recorder 1'
DEBUG:   [           75894]     << Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [           75894]     << 11
TRAFFIC: [           75894]     << 11
DEBUG:   [           75894]     >> POLL not sent
DEBUG:   [           75894]     using logical address 'Recorder 1'
DEBUG:   [           75894]     Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [           75894]     Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [           75894]     Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [           75894]     Recorder 1 (1): CEC version 1.4
DEBUG:   [           75894]     CEC::CCECClient::AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [           75894]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - trying to get the physical address via ADL
DEBUG:   [           75918]     CEC::CUSBCECAdapterCommunication::GetPhysicalAddress - ADL returned physical address 2100
DEBUG:   [           75920]     couldn't change the ackmask: the connection is closed
DEBUG:   [           75920]     Recorder 1 (1): osd name set to 'CECTester'
NOTICE:  [           75920]     CEC client registered: libCEC version = 4.0.7, client version = 4.0.7, firmware version = 65535, logical address(es) = Recorder 1 (1) , physical address: 2.1.0.0, compiled using MSVC 19.26.28806.0 (x64), features: P8_USB, P8_detect
DEBUG:   [           75920]     << Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG:   [           75920]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [           75920]     << 10
TRAFFIC: [           75920]     << 10
DEBUG:   [           75920]     >> POLL not sent
DEBUG:   [           75920]     not sending command 'set osd name': destination device 'TV' marked as not present
DEBUG:   [           75920]     << requesting power status of 'TV' (0)
DEBUG:   [           75920]     << Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [           75920]     << 10
TRAFFIC: [           75920]     << 10
DEBUG:   [           75920]     >> POLL not sent
DEBUG:   [           75920]     not sending command 'give device power status': destination device 'TV' marked as not present

This on 4.07, but that's a downgrade from 6.02 for troubleshooting purposes. It's not a direct result of firmware upgrade adapter (which I did 1,5 week ago after which it continued to work).

The USB cec adapter is connected to an AV processor (from PC). I also tested on a NUC directly connected to TV (different HDMI cable), same result.

I must say the USB CEC adpter is ancient, 8 years at least, probably older as I ordered it quickly after integration of the drivers into Kodi.

Has the adapter died or could it be something different? (bummer that i ordered a nuc cec adapter now lost at/by Post NL after I paid the customs duties).

malard, 2023-07-29:
Hey, reach out to support we can help you with a lost parcel, as for the adapter is it faulty when just connected to usb only?

trinitronx, 2024-11-16:
FWIW: I've encountered a similar [`HIGH_ERROR`][1] errors running [`cec-dpms`][2] on Linux while an [Arylic BP50][3] is connected over HDMI ARC (HDMI ARC port 1 on Vizio M601d).

```console
Nov 16 05:54:57 examplehost cec-dpms[958]: 2024-11-16, 12:54:57.185 [INFO] libcec: << Playback 1 (4) -> broadcast (F): active source (1000)
Nov 16 05:55:02 examplehost cec-dpms[958]: 2024-11-16, 12:55:02.029 [INFO] libcec: setting HDMI port to 1 on device TV (0)
Nov 16 05:55:02 examplehost cec-dpms[958]: 2024-11-16, 12:55:02.052 [INFO] libcec: >> source deactivated: Playback 1 (4)
Nov 16 05:55:02 examplehost cec-dpms[958]: 2024-11-16, 12:55:02.226 [INFO] libcec: >> source activated: Playback 1 (4)
Nov 16 05:55:02 examplehost cec-dpms[958]: 2024-11-16, 12:55:02.226 [INFO] libcec: << powering on 'TV' (0)
Nov 16 05:55:02 examplehost cec-dpms[958]: 2024-11-16, 12:55:02.297 [INFO] libcec: << Playback 1 (4) -> broadcast (F): active source (1000)
Nov 16 05:55:02 examplehost cec-dpms[958]: 2024-11-16, 12:55:02.495 [INFO] libcec: << Playback 1 (4) -> broadcast (F): active source (1000)
Nov 16 05:55:07 examplehost cec-dpms[958]: 2024-11-16, 12:55:07.567 [WARN] libcec: LOW_ERROR line:549 time:1820
Nov 16 05:55:07 examplehost cec-dpms[958]: 2024-11-16, 12:55:07.567 [WARN] libcec: TRANSMIT_FAILED_LINE
Nov 16 05:55:07 examplehost cec-dpms[958]: 2024-11-16, 12:55:07.587 [WARN] libcec: HIGH_ERROR line:218 time:1519
Nov 16 05:55:07 examplehost cec-dpms[958]: 2024-11-16, 12:55:07.597 [WARN] libcec: HIGH_ERROR line:218 time:1530

[...SNIP...]
## lots of "libcec: HIGH_ERROR" messages after this until a reboot...
```

I'm not sure what causes the issue, but usually a restart of the Linux system with the Pulse-Eight adapter resolves it for a while.  It's possible that the BP50 and/or old Vizio TV might have buggy implementation of CEC, but I have no concrete evidence of what's going wrong[^1].  I don't have any equipment such as a logic analyzer to help diagnose or listen on the HDMI lines at an electrical bit-level.

[^1]: Given the track record of the BP50's firmware updates... the problem may _likely_ be with this device.  Or, it could be something else... who knows? 🤷

[1]: https://github.com/Pulse-Eight/libcec/blob/bf5a97d7673033ef6228c63109f6baf2bdbe1a0c/src/libcec/adapter/Pulse-Eight/USBCECAdapterMessage.cpp#L111
[2]: https://github.com/manio/cec-dpms/compare/master...trinitronx:cec-dpms:master
[3]: https://forum.arylic.com/t/b50-bp50-firmware-version-update-0057-5be7aedf/4908/162

closed 2025-02-27 by github-actions[bot]