Pulse-Eight/libcec #641: No CEC replies from LG Blu Ray Player UBK90

JetA2, 2023-10-13:
Probably not a libcec problem but I'm hoping there is someone who can enlighten me on this issue.

I just bought an LG UBK90 and I wanted to check which CEC-commands it supports. I was surprised to see that no commands I send to it get any response. I have a CEC monitor device which shows me all traffic on the bus and I can see that the player sends commands to turn on the TV (not LG) when powered on, as well as broadcasts of physical address and vendor ID.

Any commands sent to the device is met with silence though. There is no ACK for a POLL so the device does not show up during a scan.

I don't get how the Simplink should work with other LG devices, unless only vendor specific codes are used for Simplink.
Does anyone have experience with CEC on LG devices?

malard, 2023-10-13:
LG Will shut off its CEC support if it finds non SimpLink devices on the bus. ☹

JetA2, 2023-10-13:
Bummer. Is there a way to make my Raspberry Pi look like a SimpLink device using libcec?

malard, 2023-10-13:
I believe its in the documentation on how to do it, @opdenkamp might be able to advise

JetA2, 2025-03-24:
Seems like libcec automatically handles LG devices.

closed 2025-03-24 by JetA2