Pulse-Eight/libcec #643: EventGhost pluging installation problem

poksh, 2023-10-19:
Installed EventGhost 0.5.0-rc6, then reexecuted libcec-6.0.2 installer with selected EventGhost pluging
Trying to add Pulse-Eight pluging into EventGhost config, but
```
Error loading plugin file: C:\ProgramData\EventGhost\plugins\PulseEight
Traceback (most recent call last) (0.5.0-rc6):
  File "C:\ProgramData\EventGhost\plugins\PulseEight\__init__.py", line 80, in <module>
    from cec_classes import UserControlCodes, CECAdapter, AdapterError # NOQA
  File "C:\ProgramData\EventGhost\plugins\PulseEight\cec_classes.py", line 53, in <module>
    from . import cec
ImportError: cannot import name cec
```
Do I need other versions of libCEC or EG?

dejiapps, 2023-10-30:
> Installed EventGhost 0.5.0-rc6, then reexecuted libcec-6.0.2 installer with enabled EventGhost pluging Trying to add Pulse-Eight pluging into EventGhost config, but
> 
> ```
> Error loading plugin file: C:\ProgramData\EventGhost\plugins\PulseEight
> Traceback (most recent call last) (0.5.0-rc6):
>   File "C:\ProgramData\EventGhost\plugins\PulseEight\__init__.py", line 80, in <module>
>     from cec_classes import UserControlCodes, CECAdapter, AdapterError # NOQA
>   File "C:\ProgramData\EventGhost\plugins\PulseEight\cec_classes.py", line 53, in <module>
>     from . import cec
> ImportError: cannot import name cec
> ```
> 
> Are these versions of libCEC and EG compatible?

I'm having the exact same issue. Did you find a solution @poksh?

poksh, 2023-10-30:
@dejiapps nope

poksh, 2023-12-17:
I tried contacting support, and I remind them once a month, but so far they only promise to contact the developer as quickly as possible

malard, 2023-12-17:
Which support are you contacting?

poksh, 2023-12-17:
@malard support on pulse-eight.com, 
I was sure that that site is related to this repository, is that correct?

malard, 2023-12-17:
Okay, but we are not eventghost, and support for these extensions is as seen, so if you need help, please ask in the eventghost forums, thanks!

poksh, 2023-12-17:
@malard oh, sorry.
But maybe you could tell me which version of EventGhost is needed for the plugin that comes with the actual libCEC installer?
I see `This file is part of the libCEC(R) library` in each .py file of plugin, so I concluded that they were tested with some version of EventGhost. I just need the version number.

MP-Munich, 2024-01-17:
I have exactly the same issue. The libCEC plugin can't be loaded, even with a completely new installed EventGhost and the default configuration.

cec-tray works fine.

I've tried the following versions:

libCEC:
- 6.0.2, released 13-Jul-2020
- 6.0.1, released 13-Jul-2020
- 6.0.0, released 10-Jul-2020
- 4.0.7, released 10-Jul-2020

EventGhost:
- 0.5.0-rc6, released 24-Dec-2018, pre-release
- 0.4.1.r1722, released 17-Feb-2016, latest official release

libCEC plugin:
1.1b, author Lars Op den Kamp

@opdenkamp
Dear Pulse-Eight team, could you please urgently solve this issue?
Obviously, the plugin files in the libCEC installation files are not working in general. Three users have been reporting the same issue.

Probably, the developer environment differs from the EventGhost default installation.

opdenkamp, 2025-02-26:
EventGhost is a third party application that hadn't been updated for many years until very recently (last release in 2018). It still uses Python 2.7 (EOL since January 2010) 32 bits (Microsoft stopped supporting installing x86 Windows). 

I have made the plugin work again in libCEC 7.0.0 and the EventGhost plugin can be installed separately now: https://github.com/Pulse-Eight/libcec/releases/download/libcec-7.0.0/libcec-eventghost-plugin-7.0.0.exe

But unfortunately I can't support third party applications, especially not when they're using an environment that's no longer supported.

closed 2025-02-26 by opdenkamp

MP-Munich, 2025-02-26:
Hello Lars,

thank you very much for updating libCEC and the EventGhost plugin. Really appreciated!

The plugin load now works and events are logged. Also, volume change and mute commands work now.
However, no other CEC command works including power on/power off. But they work using the cec-tray tool.
Any chance to fix this as well?

dejiapps, 2025-02-26:
Yes thank you. Power button functionality would be amazing please.

opdenkamp, 2025-02-26:
power control _is_ working, I've just tested it in EventGhost.
There's a bug in EventGhost that often doesn't allow you to change the name of the detected CEC adapter when setting it up, at least on my system when using the latest release of EventGhost. If I don't change the default name then it doesn't actually set up the adapter. But after the adapter has been set up power control is working.

Are you addressing the correct device(s)?

MP-Munich, 2025-02-26:
Thanks for your reply.

Directly on adding the plugin I entered "CEC" as device name:
[image]

I have only one CEC adapter, plugged in to COM4. I think the correct device is addressed since the mute commands work, as well as "Volume Up" and "Volume Down".
And I see CEC events in the EventGhost log when a device is powered on or off.

Standby the TV is working. However Power On the TV is **not** working:
[image]

[image]

When reopening the command configuration always the initial "TV" entry is selected in the dropdown menu although I selected another entry before.

"Power On All Devices" works, "Standby All Devices" not.

The "Send command to an adapter" command throws an exception:
[image]

The "Restart Adapter" command throws an exception:
[image]

dejiapps, 2025-02-27:
@opdenkamp I just tested it out again. I was able to change the name of the detected CEC adapter. Volume controls work, but power does not.

MP-Munich, 2025-02-27:
> dejiapps: I just tested it out again. I was able to change the name of the detected CEC adapter. Volume controls work, but power does not. However the logs show that the adapter detects when the T.V is manually turned off or one.

Same with me.

My environment:
- Samsung TV
- EventGhost 0.5.0-rc7
- libCEC 7.0.0

We have two different issues with the power control commands:

'1. The Standby command is refused by the TV. Same result with the cec-tray tool:
`NOTICE:              34421 << putting 'TV' (0) in standby mode
TRAFFIC:             34421 << 10:36
TRAFFIC:             34659 >> 01:00:36:01
DEBUG:               34659 >> TV (0) -> Recorder 1 (1): feature abort ( 0)`

There's no CEC Power Control option in the settings of my TV.

TV Power On via libCEC works.

'2. The major issue: The selected device in the Power command doesn't take effect and is also not saved to the configuration:
[image]

[image]

Even when clicking the "Test" button the device selection is not applied.
Hence, the power commands can't be working for any other device than the TV.

Independent from power control, there's a typo in the code:
[image]

dejiapps, 2025-02-27:
@opdenkamp @MP-Munich After digging through my T.V controls I turned on the option for the to sleep when attached device sleeps and TV to wake when attached device wakes and power options are working now 😄. Thanks for your work @opdenkamp!!!

MP-Munich, 2025-03-04:
Short update, I was able to get many functions working, including power functions for other devices than TV.
As soon as I fixed the remaining issues, I'll share the updated files with you.

MP-Munich, 2025-03-09:
Hello @opdenkamp,

two questions:

1) The volume level is first updated when I restart the adapter:
[image]
Is this a libCEC issue?

2) Does the remote key feature work with the current Samsung TV models?
When I send a key, only the code e. g. 100 for Stop is logged but the TV doesn't respond:
[image]
I tried all HDMI port numbers in the EventGhost adapter settings but no response.
Does it work on your side?

MP-Munich, 2025-03-16:
Update:

1. The volume level is first updated when I restart the adapter
Apparently, this is a know issue and a solution is ahead: https://github.com/Pulse-Eight/libcec/issues/637

2. I found out that remote key sending to my PS5 console works. Apparently, CEC remote keys are generally not supported with the Samsung TV QN modes.

I'm close to having all EventGhost plugin issues fixed and I added some new features/optimizations. Stay tuned :-)

MP-Munich, 2025-08-11:
The EventGhost plugin update is finished and has been working perfectly on my HTPC for months.
@opdenkamp: I would appreciate it if you would include this update in the next libCEC update.

**Fixed:**
- Exception on executing an action when the device is not found
- Device selection isn't saved in all device actions. The TV is always selected instead.
- Wrong volume level output
- Set volume level fails with hangup
- Remote key in "Send remote key" action cannot be changed once selected.
- "Send command to an adapter" action fails with an exception.
- Intermittent wrong disconnected/connected/power off events on AVR power on
- Unhandled exception when CEC adapter is not connected when executing actions or opening Settings
- Unhandled exception when Settings form is closed during CEC adapter loading

**Optimizations:**
- Volume level increase/decrease events. Volume level is now reported as payload, to allow event processing.
- Added device name, volume level and remote key name to action names
- Windows standby/hibernate support (adapter restart for re-establishing the adapter connection)

**New features:**
- Device power toggle action
- Device status Events on EventGhost start/adapter restart (optional setting)
- Adapter and EventGhost logging

**Known issues:**
- Volume events: Half volume steps are rounded up with audio devices with half volume steps e. g. 14.5 --> 15. No event on volume level changes from x.5 to x.0.
- Set volume is processed slowly. Reason: The CEC protocol only supports increasing/decreasing volume stepwise.
- When the audio device is powered off and the plugin is started, the mute status is "On" until the volume level is changed or the audio device is muted and unmuted.
- Some TV models report a short power on/power off cycle on CEC adapter start while the TV is powered off.

[PulseEight.zip](https://github.com/user-attachments/files/21709284/PulseEight.zip)