Pulse-Eight/libcec #644: New USB-CEC showing many transmission problems

WPettersson, 2023-10-20:
I've recently purchased a new NUC (an Intel Tiger Lake based [Geekom IT11](https://www.geekom.co.uk/geekom-mini-it11-mini-pc), and a PulseEight USB-CEC adapter to replace an Odroid C2 (RPi clone with onboard CEC). I have a Hitachi TV, a Samsung receiver, and my HTPC. The C2, using the internal CEC, was working fine with this setup where the C2 was connected to the receiver, which itself was connected to the TV.

I've run `echo "scan" | cec-client -d 31 -s` on the NUC when the USB adapter was connected to the receiver, which in turn was connected to the TV. This produced https://gist.github.com/WPettersson/ae3329165ed75c9f50ef5337820131db
I ran this inside gdb, with debugging symbols enabled, as I also get a segfault with cec-client. I will also note that this log is not always consistent.

I've also run the same command when the adapter is connected directly to the TV - no receiver connected - https://gist.github.com/WPettersson/f8235fa558bfeb15950b6377cf419a54

I also checked my C2 by running the same command, but with my C2 connected directly to the TV (note, in all earlier tests the C2 was disconnected completely and in this test, the USB adapter is disconnected completely) and in this test there are no commands that are "not acked by the controller" and the full map of everything is returned (including my receiver) - https://gist.github.com/WPettersson/8f8cdbd72d5112fe76c6c7f065646e76

Things I have tried include:

* A different HDMI cable between the adapter and the receiver. Both cables I've tried output 4k just fine to my TV, one of them was purchased just this week. 
* Either of those HDMI cables directly between the adapter and the TV, bypassing the receiver.
* Updating the firmware on the USB adapter - the firmware update runs successfully and reports v12 being uploaded.
* Swapping out the tiny cable between the NUC and the USB adapter - this has no effect that I have noticed, the logs still look the same

Comparing the logs between the NUC+USB adapter and the C2, it seems the USB adapter is reporting a lot of `command 'CEC transmission' was not acked by the controller` for various commands. I don't know enough about CEC to know if this is a failure between the NUC and the USB adapter, or between the USB adapter and the TV. There's also a lot of `CEC transmission - received response - TRANSMIT_FAILED_TIMEOUT_LINE`

Is there something else I could've missed that would make CEC not work?

For the record, while CEC isn't working, the actual HDMI seems to be working fine (including 4k display) on any of the cables even with the USB adapter connected, so it's just CEC that is giving me issues.

WPettersson, 2023-12-26:
I've decided to have another poke at this and can report the following:

The USB-CEC still has the same failure if I set my NUC to only output 1280x1024 @ 60Hz.

The USB-CEC does "work" correctly to identify the TV and receiver if I disconnect the HDMI cable from the NUC to the USB-CEC.

The USB-CEC breaks again if I connect a HDMI cable from the NUC directly to the TV. In this setup, I have a cable from NUC to TV, a cable from TV to receiver, and a third cable from USB-CEC to TV. There is no cable connected to the "PC" input of the USB-CEC.

malard, 2023-12-26:
This sounds like the NUC is still pulling the CEC wire low, CEC is a common bus, so all the ports on the TV are wired together

WPettersson, 2023-12-26:
Thanks for that info, you precisely nailed the issue. I did the electrical-tape-on-pin-13 trick and CEC is now working both with cec-client and with Kodi :)

Is it worth investigating the crash that results when the NUC does pull pin 13 down? The backtrace seems to indicate a problem in the SAFE_DELETE at https://github.com/Pulse-Eight/libcec/blob/bf5a97d7673033ef6228c63109f6baf2bdbe1a0c/src/libcec/CECProcessor.cpp#L137 but this is only reproducible for me when I leave pin 13 from the NUC connected.

malard, 2023-12-26:
If someone wants to, they can. But this is a hardware induced fault

WPettersson, 2023-12-30:
Ok, closing as complete

closed 2023-12-30 by WPettersson