Pulse-Eight/libcec #646: Cannot send OSD key presses to TV

kongomongo, 2023-10-24:
Hi there,

I have a running implementation off of CecSharpClient. Pretty much everything works. Now I wanted to implement two features: Add a global hotkey so that Ctrl-Win-Plus and Ctrl-Win-Minus would send .VolumeUp() and .VolumeDown() commands. This works.

Now sometimes my LG OLED comes on with nagging about "TV will shut off in 5 minutes" and if I dont have the remote but just they keyboard i cant just click it away.

So i simply wanted to send a Enter/Select to the TV from Code triggered via a keyboard hotkey. No matter what I do this fails.

So volumeup sends 15:44:41

And my keypress sends 10:44:00.

Both looks ago, I even tried all available combinations of logicaladdresses and also Enter / Select . But nothing happens on the LG.

Am I generally doing it wrong? Can the TV receive OSD-steering commands?

Anyone done this?

MANY thanks in advance.

kongomongo, 2023-12-05:
Is this generally even possible to send a keypress to the TV? Does it generally work on LG? Does anyone have any experience?

kongomongo, 2024-01-22:
Hrm, is it wrong to post this as an issue? Is there a better place?

opdenkamp, 2025-02-27:
it's not supported by the CEC 1.4b protocol unfortunately. some TVs do support it, but it's not in the spec.

closed 2025-02-27 by opdenkamp