Pulse-Eight/libcec #650: Was working before upgrade?

Jibun-no-Kage, 2023-11-29:
After most recent PiOS update, cec-client was working, no fails instantly,

Linux eccentric 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64
Last login: Wed Nov 29 04:10:26 2023
\# cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect: FAILED

\# echo 'scan'|cec-client -s -d 1
autodetect FAILED

Any suggestions?

\# uname -a
Linux eccentric 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

\# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Jibun-no-Kage, 2025-02-26:
Cannot test with CEC 7.0... not available...
```
Get:1 http://deb.debian.org/debian bookworm/main arm64 libp8-platform2 arm64 2.1.0.1+dfsg1-4 [20.9 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libcec6 arm64 6.0.2-5+rpt2 [188 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 cec-utils arm64 6.0.2-5+rpt2 [30.0 kB]
```

Jibun-no-Kage, 2025-02-26:
Seems that original scan error fixed overall with Debian 12 based PiOS release?
```
# echo 'scan' | cec-client -d 1
log level set to 1
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect:
 path:     /sys/devices/platform/soc/3f902000.hdmi/cec0
 com port: /dev/cec0

opening a connection to the CEC adapter...
waiting for input
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```
But the active source issue remains.

opdenkamp, 2025-02-27:
this would be something for whoever packaged/built this because libCEC didn't change at that point in time and it still works fine on other Linux distributions that are running on the Pi.

closed 2025-02-27 by opdenkamp