Pulse-Eight/libcec #651: adapter Open() leaves defunct process if pi booted when hdmi is unplugged

jackhamby, 2023-12-06:
Raspberry Pi Compute Module 4 Rev 1.0
Ubuntu 22.10
libCEC version 6.0.2

1. power down pi
2. unplug hdmi from the pi
3. power on pi
4. start cec client (it should hang)
5. try killing cec-client (it should leave <defunct> process)

When i boot the pi up while hdmi is not plugged in and try to use cec, the libcec process hangs and cannot be killed. There is a [timeout on the adapter.Open()](https://github.com/Pulse-Eight/libcec/blob/master/src/libcec/LibCEC.cpp#L76) that isn't honored.

can reproduce using cec-client
[image]

try to SIGINT and SIGKILL the cec-client...
[image]

try to start and stop again leaves defunct process
[image]

plugging the hdmi in while its opening doesnt change the behavior, it never recovers.
only option is to restart the pi.
is this a known issue with the libcec library or is it a hardware level thing?
im looking to know how to recover from this state, when libcec is opened without the pi being physically connected to cec bus. Or if there is another way we can detect that hdmi is unplugged from the pi before we try to open the adapter?

popcornmix, 2023-12-06:
Possibly this is an Ubuntu issue.
libcec can be built two ways for the pi:
`-DHAVE_RPI_API=1` build for the deprecated firmware display stack
`-DHAVE_LINUX_API=1` build for the recommended linux side display stack

Ubuntu (and any recent OS) will be using the linux side display stack (KMS/DRM).
I have a suspicion (`com port: RPI`) this has been built wrongly.

You could try building this locally for linux, and if it fixes it, then inform Ubuntu devs.

jackhamby, 2023-12-21:
@popcornmix thank you for the response.
building in docker like so,
[image]

we still get the same issue. cannot kill libcec process, any attempts to results in D state for the process (uninterruptable sleep). is there anything else you can recommend trying?

jackhamby, 2024-01-22:
after digging into it more, we realized were still pulling the RPI driver. once we got libcec to point to the linux driver this locking up stopped happening.  thanks @popcornmix for pointing us in the right direction

closed 2024-01-22 by jackhamby