Pulse-Eight/libcec #652: Using libcec as proxy for AV receiver - How do I put audio in standby when handled by libCEC?

ballle98, 2024-01-04:
working on a CEC to IR bridge here https://github.com/ballle98/cec-lirc

If the TV is off I turn the audio off but if the TV asks the power status libCEC always reports on.  Is there a way to get libCEC to report "standby" for a device it handles?

```
CECCommand: opcode 36 0 -> f
turnAudioOff CECCommand: lirc_send_one KEY_SUSPEND
[01/04/24 10:06:04.0870] LOG16 TV (0): power status changed from 'on' to 'standby'
[01/04/24 10:06:04.0870] LOG16 >> TV (0) -> Broadcast (F): standby (36)
[01/04/24 10:06:04.0870] LOG4 << putting 'Audio' (5) in standby mode
[01/04/24 10:06:04.0870] LOG16 not sending command 'standby': destination device 'Audio' marked as handled by libCEC
[01/04/24 10:06:04.0870] LOG16 not sending command 'system audio mode request': destination device 'Audio' marked as handled by libCEC
[01/04/24 10:06:04.0895] LOG8 >> 05:a4:0a
CECCommand: opcode a4 0 -> 5
[01/04/24 10:06:04.0896] LOG16 sending abort with opcode a4 and reason 'unrecognised opcode' to TV
[01/04/24 10:06:04.0896] LOG16 << transmitting abort message
[01/04/24 10:06:04.0896] LOG8 << 50:00:a4:00
[01/04/24 10:06:04.0896] LOG16 >> TV (0) -> Audio (5): UNKNOWN (A4)
[01/04/24 10:06:05.0072] LOG8 >> 05:9f
CECCommand: opcode 9f 0 -> 5
[01/04/24 10:06:05.0072] LOG16 << Audio (5) -> TV (0): cec version 1.3a
[01/04/24 10:06:05.0072] LOG8 << 50:9e:04
[01/04/24 10:06:05.0072] LOG16 >> TV (0) -> Audio (5): get cec version (9F)
[01/04/24 10:06:05.0257] LOG8 >> 05:8c
CECCommand: opcode 8c 0 -> 5
[01/04/24 10:06:05.0257] LOG16 << Audio (5) -> TV (0): vendor id Pulse Eight (1582)
[01/04/24 10:06:05.0257] LOG8 << 5f:87:00:15:82
[01/04/24 10:06:05.0257] LOG16 >> TV (0) -> Audio (5): give device vendor id (8C)
[01/04/24 10:06:05.0459] LOG8 >> 05:8f
CECCommand: opcode 8f 0 -> 5
Power Status(Audio): on
turnAudioOn: lirc_send_one KEY_POWER
[01/04/24 10:06:05.0527] LOG16 << Audio (5) -> TV (0): on
[01/04/24 10:06:05.0527] LOG8 << 50:90:00
[01/04/24 10:06:05.0527] LOG16 >> TV (0) -> Audio (5): give device power status (8F)
[01/04/24 10:06:05.0527] LOG16 not sending command 'system audio mode request': destination device 'Audio' marked as handled by libCEC
[01/04/24 10:06:05.0527] LOG16 'Audio' (5) is already 'on'
[01/04/24 10:06:05.0612] LOG8 >> 05:8c
CECCommand: opcode 8c 0 -> 5
[01/04/24 10:06:05.0612] LOG16 << Audio (5) -> TV (0): vendor id Pulse Eight (1582)
[01/04/24 10:06:05.0612] LOG8 << 5f:87:00:15:82
[01/04/24 10:06:05.0612] LOG16 >> TV (0) -> Audio (5): give device vendor id (8C)
[01/04/24 10:06:05.0813] LOG8 >> 05:8f
CECCommand: opcode 8f 0 -> 5
Power Status(Audio): on
turnAudioOn: lirc_send_one KEY_POWER
```

malard, 2024-01-04:
The adapter will do so (I think) if the connection is severed, otherwise its assumed on, the code is open so change it as you wish 😊

ballle98, 2024-01-04:
I'm using a Raspberry Pi not an external adapter.  Not sure which code I would change.

malard, 2024-01-04:
The source code of libcec and recompile it

closed 2025-02-26 by ballle98