Pulse-Eight/libcec #653: echo "scan" CEC failed on raspberry 4

Louishattiger, 2024-01-10:
Background:
Sending commands to my computer or smart tv via an hdmi cable from a raspberry 4 (volume up/down, play/pause, mute/demute,...)

I first have this problem:

sudo echo 'scan' | cec-client -s -d 1
opening a connection to the CEC adapter...
ERROR:   [             675]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=64
 CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

Then i followed the issue #597 and I changed dtoverlay=vc4-kms-v3d to dtoverlay=vc4-fkms-v3d in config.txt (and reboot the rasp).
Now it seems that I take a step backwards:

echo "scan" | cec-client -s -d 1
autodetect FAILED

sudo cec-client -i
libCEC version: 6.0.2, compiled on Linux ... , features: P8_USB, DRM, P8_detect, randr, Exynos, Linux, AOCEC

uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

I'm a bit lost :(

Thank you in advance for your help

wallagib, 2024-03-01:
I'm seeing this issue too. Right now, I'm struggling to build from source because, even though I'm following the RPi guide, it requires `RPI_BCM_HOST` to be set, and this apparently doesn't exist in newer Raspbians.

wallagib, 2024-03-01:
This comment also has some tips for building, but I didn't have much luck the first time around: https://github.com/Pulse-Eight/libcec/issues/470#issuecomment-629702541

opdenkamp, 2025-02-28:
I've updated the documentation to point out that the Linux Kernel driver should be used instead of the Raspberry Pi ones now.

closed 2025-02-28 by opdenkamp