Pulse-Eight/libcec #661: When will LibCecSharp be upgraded to support .NET 8.0?

SmartLifeSolutions, 2024-08-29:
We have an application that runs on Windows 10/11 Pro that currently uses .NET Framework 4.8.1 and everything has been working well for the last few years since we launched our product that uses the USB-CEC. However, we are now looking to integrate MQTT into our app and the dll we are using for MQTT only supports .Net 6.0, 7.0 and 8.0. 

We would like to move our application to .NET 8.0, but neither LibCecSharp nor LibCecSharpCore are compatible with .NET 8.0, and can't be referenced in the Visual  Studio 2022 project.

Is there a possibility that Pulse Eight will upgrade LibCecSharp to work with newer versions of .NET, and not just 
.NET Framework and .NET Core? If not, can someone provide me with instructions on how to upgrade LibCecSharp myself?

olegsavelos, 2024-10-03:
@malard We are also interested in developing an cross platform .net wrapper, possibly with AOT support, the question is whether it would be possible for your team to assist us at least at consulting level ?

malard, 2024-10-03:
Sure, what info do you need?

olegsavelos, 2024-10-03:
@malard I suppose some clarifications from the native library dev, I still need to gather our questions and that will only happen once we commit to the project, having positive response for you is sure encouraging :)

Will be getting back to you on this.

szv, 2024-12-05:
@SmartLifeSolutions I just tried _LibCecSharp_ with a .NET 9 console and WPF application. It works in both cases.

It is just important to know that you have to use the "netfx" version instead of the "netcore" version (which is very misleading in my opinion).

I downloaded the [LibCec 6.0.2](https://github.com/Pulse-Eight/libcec/releases/tag/libcec-6.0.2) installer and installed _libCEC for .Net Framework_.
[image]

After that I added a reference to the `LibCecSharp.dll` in my project file (in this case a class library) like that:
```xml
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>net9.0</TargetFramework>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
  </PropertyGroup>

  <ItemGroup>
    <Reference Include="LibCecSharp">
      <HintPath>..\..\..\..\..\Program Files (x86)\Pulse-Eight\USB-CEC Adapter\x64\netfx\LibCecSharp.dll</HintPath>
      <Private>true</Private>
    </Reference>
  </ItemGroup>

</Project>
```

After that I could use CecSharp in my .NET 9 application like in this sample:

```C#
using CecLibrary.Exceptions;
using CecSharp;

namespace CecLibrary;

public class CecService
{
    private readonly LibCecSharp _cecLib;

    private readonly CecCallbacks _callbacks = new();

    public CecService()
    {
        var config = new LibCECConfiguration()
        {
            ActivateSource = true,
            AdapterType = CecAdapterType.PulseEightExternal,
            AutoPowerOn = BoolSetting.Enabled,
            ClientVersion = LibCECConfiguration.CurrentVersion,
            GetSettingsFromROM = true,
            PowerOffOnStandby = true,
            TvVendor = CecVendorId.Samsung,
        };
        _cecLib = new LibCecSharp(_callbacks, config);
        _cecLib.InitVideoStandalone();
    }

    public void Start()
    {
        var adapter = _cecLib
            .FindAdapters(string.Empty)
            .FirstOrDefault() ?? throw new Exception("No adapter found");
       
        _cecLib.Open(adapter.ComPort, 10000);
    }
}
```

```C#
using CecSharp;

namespace CecLibrary;

internal class CecCallbacks : CecCallbackMethods
{
    public override int ReceiveKeypress(CecKeypress key)
    {
        Console.WriteLine($"Key {key.Keycode} was pressed for {key.Duration}");
        return 0;
    }

    public override int ReceiveCommand(CecCommand command)
    {
        Console.WriteLine($"Command {command.Opcode} was received");
        return 0;
    }

    public override int ReceiveMenuStateChange(CecMenuState newVal)
    {
        Console.WriteLine($"Menu state changed to {newVal}");
        return 0;
    }

    public override void SourceActivated(CecLogicalAddress logicalAddress, bool activated)
    {
        Console.WriteLine($"Source {logicalAddress} was {(activated ? "activated" : "deactivated")}");
    }

    public override int ReceiveLogMessage(CecLogMessage message)
    {
        Console.WriteLine($"Log message: {message.Level} - {message.Message}");
        return 0;
    }
}

```

opdenkamp, 2025-02-26:
libCEC 7.0.0 has bumped the .Net Core library to .Net 8 (Microsoft have changed the name of the framework)

closed 2025-02-26 by opdenkamp