Pulse-Eight/libcec #662: Cant get this to work on Linux Mint

Asdf1qaz, 2024-09-09:
Kind of a noob to code etc, but I have tried to install this and cant get Kodi to recognise the CEC adapter on Linux Mint.
I have done:

To install P8: 

apt-get update
apt-get install cmake build-essential
git clone https://github.com/Pulse-Eight/platform.git
mkdir platform/build
cd platform/build
cmake ..
make -j4
sudo make install
sudo ldconfig

And then to install LibreCEC:

apt-get update
apt-get install cmake libudev-dev libxrandr-dev python-dev swig
git clone https://github.com/Pulse-Eight/libcec.git
mkdir libcec/build
cd libcec/build
cmake ..
make -j4
sudo make install
sudo ldconfig

I dont think I got any errors or anything, but Kodi still wont recognise it and on boot, is coming up with:

Couldn't initialise the CEC adapter. Please check your settings.

What settings do I need to change?
Is there anything else that I need to do?
I did see "https://github.com/Pulse-Eight/libcec/blob/master/docs/README.debian.md" But im not sure what to do with that? When/where do I need to run these?
I cant find anything, but is there something that I need to do within Kodi to set it up?
Do I need to enable a service to boot it?

For such an expensive little box, I was expecting the instructions to be a bit clearer to an average pleb

malard, 2024-09-09:
it's natively supported in Kodi, so use a standard package rather than building it yourself. "average citizen (pleb)" should not be trying to make all their own software and just use an off the shelf package

Cheers!

closed 2024-09-09 by malard

Asdf1qaz, 2024-09-10:
Where can I find the package? I cant find it in the add on browser?

Asdf1qaz, 2024-09-10:
Also, maybe you could add a sentence or two to the info page stating that, as without you telling me that directly, how was I meant to know?

malard, 2024-09-10:
It's natively built into kodi. You don't need any packages, did you manually compile Kodi or just install it via a package manager?