Pulse-Eight/libcec #668: libcec prevents creation of new /dev/cec0 device

jeffgriffin, 2025-01-02:
On my Intel NUC; /dev/cec0 gets destroyed, then created again when my receiver goes into standby mode.  I have tested this with this watch command, while no libcec clients are running:
`watch -d -t -g eval '[ -e /dev/cec0 ] && stat /dev/cec0 2>/dev/null'`
It happens quickly, but the file is removed, then replaced with a new device file with a new timestamp.

If a libcec client is running (whether via python-cec or just cec-client monitoring), the new device will fail to be created at /dev/cec0.  It will instead appear at /dev/cec1, which will not be discoverable by python-cec or cec-client.

opdenkamp, 2025-02-27:
This is really a problem of your NUC's implementation and not something libCEC can prevent. If the device becomes inaccessible then libCEC will retry and eventually exit, but that's not instant and it will keep the node locked while it's still running.

jeffgriffin, 2025-03-14:
Isn't the creation of a new node an opportunity to test validity of an existing lock?

shelvacu, 2025-11-03:
I ran into this with simply unplugging and replugging a displayport to hdmi adapter

Why is the path hardcoded as `/dev/cec0`? Does that mean the library can't handle more than one cec connection? It should search for all `/dev/cec*` ideally

shelvacu, 2026-04-29:
still relevant https://github.com/Pulse-Eight/libcec/blob/c7c4c82dc171c49decfbfe5d2959706d3e2ea47c/include/cectypes.h#L292

endrift, 2026-04-29:
We're really adding a stalebot to a repo that gets less than one commit a month? Really?