Pulse-Eight/libcec #669: Build libcec on Windows ARM64

ddscentral, 2025-01-13:
In it's current state, libcec cannot be built for Windows ARM64, but there doesn't appear to be any code in the library that's specific to the x86 platform. With some minor changes to the build scripts and headers, the native library builds fine. 
The .NET library isn't buildable as AFAIK the .NET Core SDK isn't available for ARM64

I have created forks of the master branch as well as "platform" and "support" submodules with the necessary changes.
Would be good to have someone review these changes before submitting PRs so I can make any corrections if needed.

https://github.com/ddscentral/libcec-win-arm64
https://github.com/ddscentral/llibcec-platform-win-arm64
https://github.com/ddscentral/libcec-support-win-arm64

garbear, 2025-01-14:
I think it's time for PRs, I went through the work and there were no surprises. It looks like a typical platform port. Good work!

ddscentral, 2025-01-15:
Created PRs for main repo as well as "platform" and "libcec-support".

https://github.com/Pulse-Eight/libcec/pull/670
https://github.com/Pulse-Eight/platform/pull/50
https://github.com/Pulse-Eight/libcec-support/pull/4

opdenkamp, 2025-02-14:
thanks! I've merged these in and this will be included in the next libCEC release. working on that one now

closed 2025-02-14 by opdenkamp