Pulse-Eight/libcec #673: Compiled from source on PiOS natively, no cross compile, and cannot complete a scan?  RPI serial device not defined as once was?

Jibun-no-Kage, 2025-02-26:
Compiled from source on PiOS natively, no cross compile, and cannot complete a scan?  RPI serial device not defined as once was?

echo 'scan' | cec-client -d 1

```
log level set to 1
No device type given. Using 'recording device'
CEC Parser created - libCEC version 7.0.0
no serial port given. trying autodetect: FAILED
```
echo 'scan' | cec-client -d 1 -s RPI
```
log level set to 1
opening a connection to the CEC adapter...
ERROR:   [            3506]     error opening serial port 'RPI': No such file or directory
ERROR:   [            3506]     could not open a connection (try 1)
ERROR:   [            6342]     error opening serial port 'RPI': No such file or directory
ERROR:   [            6342]     could not open a connection (try 2)
ERROR:   [            8232]     error opening serial port 'RPI': No such file or directory
ERROR:   [            8232]     could not open a connection (try 3)
^Csignal caught: 2 - exiting
ERROR:   [            9492]     error opening serial port 'RPI': No such file or directory
ERROR:   [            9492]     could not open a connection (try 4)
unable to open the device on port RPI
```
echo 'scan' | cec-client -d 1 -s 1
```
[Results in endless loop, cannot break out, had to abort/kill session]
```

Jibun-no-Kage, 2025-02-26:
No devices found...
```
cec-client -l
libCEC version: 7.0.0, git revision: libcec-6.0.2+82-439710e, compiled on 2025-02-26 22:33:14 by root@eccentric.dachshund-digital.org on Linux Linux (Linux), features: P8_USB, DRM, P8_detect, randr
Found devices: NONE
```
At least version 4 could see devices before connected to a Raspberry Pi.

Jibun-no-Kage, 2025-02-26:
Well there is no doubt that the device exists...
```
find /dev | grep cec
/dev/cec0
```

opdenkamp, 2025-02-27:
your build doesn't include support for the Raspberry Pi's interface. Check your cmake output.
Are `RPI_LIB_DIR` and `RPI_INCLUDE_DIR` defined in your cmake command and pointing to the correct directories?

Jibun-no-Kage, 2025-02-28:
Followed the documented directions specific to the Pi environment direct compile yes.  If the correct LIB_DIR and INCLUDE_DIR were wrong the cmake should have explicitly failed when compiling in the Pi OS on Pi hardware no?

closed 2025-02-28 by opdenkamp

opdenkamp, 2025-02-28:
I've just checked this in more detail. On a recent Raspberry Pi that's using the `v4-kms-3d` dtoverlay, you have to use the standard Linux kernel API to use CEC, not the Raspberry Pi specific one. Rebuild libCEC with these options, then it'll work on the Pi: `cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0`

I'll update the documentation to point this out.

opdenkamp, 2025-02-28:
see https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md#compilation-using-the-linux-kernel-driver