Pulse-Eight/libcec #685: EventGhost plugin: Remote control input

MP-Munich, 2025-03-28:
In the source code of the EventGhost plugin I see that the HDMI port setting is read in but not actively used.
Is it planned to add support for remote control input, or can this EventGhost setting be removed?

opdenkamp, 2025-06-19:
> I set the EventGhost plugin AVR HDMI port number setting to from 1 to 6 but remote control keys are not output to EventGhost. Any experience with Denon AVRs? How to confirm the HDMI port number? Can we see this in the cec-console output?

The correct value is auto-detected on most setups. If not, then you can check the labels on the back of the AVR

> Is remote control key receiving from **TV** also supported? If yes, does the CEC device need to be connected to the TV instead of the AVR?

This is the only fully supported way for remote controls, from TV to source device (CEC adapter in this case). AVR remotes will only pass volume control key presses in most cases.

You can connect the adapter in every spot in the HDMI chain, but what matters for setting up the correct HDMI port number in EventGhost is where you've connected the HDMI output of your PC. That's the number to set up.

closed 2025-06-19 by opdenkamp

MP-Munich, 2025-07-28:
Thanks for your reply.

If the CEC adapter is connected to the AVR and we like to receive TV remote control keys, is the HDMI port number of the AVR or the TV to be entered to the configuration?
With this kind of chain/setup, can keys of **both** devices (ARV and TVs) be received?