Pulse-Eight/libcec #688: Instructions in linux install outdated as python-dev is no longer available

Asdf1qaz, 2025-04-08:
The instructions in docs/README.linux.md are outdated and no longer work
Please update them so we can still install.

Package python-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-dev-is-python3

E: Package 'python-dev' has no installation candidate

Can we use python-dev-is-python3?

Asdf1qaz, 2025-05-16:
Guys?

Asdf1qaz, 2025-06-05:
Updates please

Asdf1qaz, 2025-06-10:
Updates please

malard, 2025-06-10:
Have you tried to use python-dev-is-python3 instead? This is not the customer support portal, if you have an issue with a Pulse-Eight hardware product, email cs@pulse-eight.com not here, this area is not monitored by our support team

closed 2025-06-19 by opdenkamp