Pulse-Eight/libcec #691: ARC support

drolevar, 2025-06-03:
Is it somehow possible to make USB CEC adapter act like an HDMI ARC receiver?

I've managed to make a Python script with the newest libcec version 7.0.0 which handles ARC start request, but TV immediately responds with ARC Terminated, which probably means the adapter is unable to receive ARC data.

Is it something that would be possible to solve on the libcec level, or it's necessary to update the firmware of the adapter?

malard, 2025-09-01:
No it's not possible. CEC is only the comms, the data is on a different wire and you need an i2s chip etc to deal with this

closed 2025-09-01 by malard

malard, 2025-09-01:
We sell other products that do this, feel free to email sales.

Thanks